ros2/rosbag2 #632: Switch the zstd_vendor PATCH_COMMAND to use "git apply"

clalancette, 2021-02-03:
In https://github.com/ros2/rosbag2/issues/630, the reporter ran into trouble applying the patch from zstd_vendor on Windows.  A long-standing issue on Windows is that the `patch.exe` binary requires Administrator permissions to run.  In most of the rest of the ROS 2 ecosystem, we've switched to using `git apply` to apply patches instead, which doesn't require Administrator permissions.  You can see an example of how we use it at https://github.com/ament/uncrustify_vendor/blob/8e622cf0f996eb1e15b58b490cd7bed360a8ceac/CMakeLists.txt#L57

We should change the zstd_vendor CMakeLists.txt to do something similar.

emersonknapp, 2021-02-03:
Thanks for reporting this - seems sensible. There are a few other vendor packages with patch commands, just noting it here for reference: https://github.com/search?q=org%3Aros2+PATCH_COMMAND&type=code. I think that `rosbag2_bag_v2` may need an update as well, when we grab this ticket. `tinyxml_vendor` seems to use `git apply` also - https://github.com/ros2/tinyxml_vendor/blob/f97ba1255b1ac15695285864d6482525119c4e03/CMakeLists.txt#L31

dimi575, 2021-08-12:
Git installs a patch.exe binary that does not require Admin rights. I made a pull request to change the Windows build instructions in the documentation to include this in the PATH environment variable, fixing the build failure.
https://github.com/ros2/ros2_documentation/pull/1815

clalancette, 2021-08-12:
We've switched almost all other vendor packages to use `git apply`.  I'd rather be consistent here and use that where we can.

christophebedard, 2021-08-12:
See #846

PR #846 (2021-08-13): Switch to using 'git apply' for zstd_vendor patches
Closes #632

I checked the resulting external project build directory (`build/zstd_vendor/zstd-1.4.4-prefix/src/zstd-1.4.4`) and patches seem to be applied correctly.
closed 2021-08-13 by emersonknapp