ros2/rosbag2 #642: benchmark tool reports higher than 100% recorded percentage

y-okumura-isp, 2021-02-09:
## Description

`rosbag2_performance_benchmarking` reports higher than 100% recorded percentage.

### Expected Behavior
`<= 100%`

### Actual Behavior

When we run test with transport mode, we got the following report.

```
        Recorded messages for different caches and storage config:
                storage config: default:
                        cache [bytes]: 100,000,000: 100.04% recorded
```

## To Reproduce

Change `no_transport: False`.
https://github.com/ros2/rosbag2/blob/master/rosbag2_performance/rosbag2_performance_benchmarking/config/benchmarks/test.yaml#L8

And follow the steps in https://github.com/ros2/rosbag2/tree/master/rosbag2_performance/rosbag2_performance_benchmarking.

```
$ ros2 launch rosbag2_performance_benchmarking benchmark_launch.py benchmark:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/benchmarks/test.yaml producers:=`ros2 pkg prefix rosbag2_performance_benchmarking`/share/rosbag2_performance_benchmarking/config/producers/mixed_110Mbs.yaml

$ scripts/report_gen.py -i <BENCHMARK_RESULT_DIR>
```

## System (please complete the following information)
 - OS: ubuntu 20.04
 - ROS 2 Distro: rolling
 - Version: 7548a04c3b5839ed839ab3ae7c5d25f0d522807d

## Additional context

- `report_gen.py` looks `total_recorded_count` in `results.csv`
- `total_recorded_count` is from `message_count` in `metadata.yaml`
- `message_count` contains ALL topics including `/rosout` or `/parameter_events` and so on.

It looks OK to filter message counts in `result_utils.cpp`.

adamdbrw, 2021-02-10:
The reason is catching /rosout. Currently two ways to fix it:
#634 employs filtering by regex to only count topics from benchmarking tool
#643 extends parsing of metadata yaml file to only count with a given namespace

y-okumura-isp, 2021-02-12:
As I closed PR #643, I also close this issues.

closed 2021-02-12 by y-okumura-isp