ros2/rosbag2 #647: Improving performance of should_split_bagfile()

adamdbrw, 2021-02-11:
## Description
Bag performance suffers due to should_split_bagfile() checking file size for each incoming message. This is a simple disk operation but adds up very quickly, especially when lots of messages are going through (e. g. high-frequency publishers and/or many publishers).

### Expected Behavior
Determine whether to split the bag has a negligible cost compared to the cost of writing the message itself.

### Actual Behavior
Checking for the split condition is dominating the main thread in some cases (credit to @eboasson for finding out with a macOS datapoint).

## To Reproduce
Use the rosbag2_performance_benchmarking package. Set max bag size to a non-zero value. Make sure to run benchmarks setup with high message throughput (high frequencies and/or plenty of publishers). Smaller message size with the same MB/s would show a more pronounced effect.

## Additional context
Bag splitting itself could also be improved: https://github.com/ros2/rosbag2/issues/640

emersonknapp, 2021-02-12:
Good catch - this performance profiling is really helping rosbag2 a lot. I'm sure we can find an implementation that has negligible performance impact.

adamdbrw, 2021-02-12:
If we ignore storage overhead, we already look at each message size because of the cache limit. We could use this (cumulative) size to track when to split the bagfile.

emersonknapp, 2021-02-12:
Do you know how much storage overhead there is? Is it constant, or a constant multiplier? If so then we could query that value once from the storage implementation - and like you said, use a cumulative counter without querying for each message.

pijaro, 2021-02-23:
I did some quick benchmarks and it is more like a constant multipler of about ~100.11% for sqlite3:
- for smaller bag size: 1GB - it varies a bit depends on total number of messages - 100.38% for 10k (filesize 1003.762MB), 100.11% for 2k (filesize 1001.116MB) and 100.15% for 1k (filesize 1001.505MB),
- for larger bag size: 5GB - it is much flatter with about 100.111% for 10k (filesize 5005.533MB), 100.108% for 2k (filesize 5005.390MB) and 100.107% for 1k (filesize 5005.357MB) messages,
- for the same setup (producers & benchmark) db filesize is constant.

These tests were conduced with 1 and 10 publishers. Results were identical (since topic names are stored in db after all, probably rising publishers even more would change anything, didn't test that).

MichaelOrlov, 2023-12-14:
Update note: 
This is not an issue anymore with the MCAP file format since MCAP writer does track the cumulative size of the written data internally and doesn't do file operation when requesting the `get_bagfile_size()` API call.
However, for SQLite - this is still an issue and sort of suboptimal implementation since we are calling `(rcpputils::fs::path{get_relative_file_path()}).file_size()` per each message write if `storage_options.max_bagfile_size` not equal to **0**.
But on another hand, the real size of the written SQLite3 file could be and will be different from what we could count on the upper layer of written bytes due to the indexes in the database.

closed 2024-01-12 by MichaelOrlov