ros2/rosbag2 #649: Include Paths are Messed Up in _Writer.cpp and Others

cameronschloer, 2021-02-12:
I just cloned the repository and went in to look at _Writer.cpp and it said it couldn't find the files that it was trying to include. Upon further inspection, it looks like the paths are incorrect and need to be fixed. Am I missing something obvious or is this something that needs to be changed?

emersonknapp, 2021-02-12:
The code builds on ci.ros2.org, on the Action CI https://github.com/ros2/rosbag2/actions, and on my developer system.

It would be better to use the issue template that is provided in this repository for Bug Report when reporting issues. It prompts you for the sort of information that could help someone understand your problem. That information is things like:
* what system are you running on?
* what commands are you running?
* what are you expecting to happen?
* what output is actually happening? (copy-paste, don't paraphrase)

However, this does seem like a more general ROS development question and is probably better suited to answers.ros.org

cameronschloer, 2021-02-12:
Oh my bad! I'm still pretty new to ROS and I wasn't aware that I should take that path. I will do that for now on. I'll start with that. Sorry to bother!

closed 2021-02-12 by cameronschloer