ros2/rosbag2 #650: Can't build rosbag2_storage master branch

Kaju-Bubanja, 2021-02-12:
## Description
I have a source checkout of ros2 master branches from [here](https://index.ros.org/doc/ros2/Installation/Maintaining-a-Source-Checkout/#maintainingsource). When I try to build rosbag2_transport master branch I get following error message
```

--- stderr: rosbag2_storage                                                                                                                                                        
/home/usr/ros2_foxy/src/ros2/rosbag2/rosbag2_storage/src/rosbag2_storage/metadata_io.cpp: In member function ‘virtual rosbag2_storage::BagMetadata rosbag2_storage::MetadataIo::read_metadata(const string&)’:
/home/usr/ros2_foxy/src/ros2/rosbag2/rosbag2_storage/src/rosbag2_storage/metadata_io.cpp:236:53: error: could not convert ‘& metadata.rosbag2_storage::BagMetadata::bag_size’ from ‘uint64_t*’ {aka ‘long unsigned int*’} to ‘rcutils_allocator_t’
  236 |       rcutils_calculate_directory_size(uri.c_str(), &metadata.bag_size, allocator))
      |                                                     ^~~~~~~~~~~~~~~~~~
      |                                                     |
      |                                                     uint64_t* {aka long unsigned int*}
make[2]: *** [CMakeFiles/rosbag2_storage.dir/build.make:63: CMakeFiles/rosbag2_storage.dir/src/rosbag2_storage/metadata_io.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:129: CMakeFiles/rosbag2_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_storage [8.05s, exited with code 2]
```

### Expected Behavior
I would expect it to build without error

### Actual Behavior
See above error message

## To Reproduce
1. colcon build --symlink-install rosbag2_storage

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: master checkout built by source
 - Version: master

## Additional context
I can build foxy without a problem, but not master

emersonknapp, 2021-02-12:
Have you built the other rosbag2 dependency packages as well (`colcon build --packages-up-to rosbag2_storage`), or are you trying to do an isolated single-package build against some other installed binaries?

Kaju-Bubanja, 2021-02-15:
I thought colcon building a package automatically builds the required packages first? I tried above command but still get the same error. I don't want to delete the whole workspace again, because it takes a long time to rebuild it. But I deleted the rosbag2_storage folder in install and build. `Git pull` gives `Already up to date.` `Git status` gives `On branch master
Your branch is up to date with 'origin/master'.` `git remote -v` gives: `origin	https://github.com/ros2/rosbag2.git (fetch)
origin	https://github.com/ros2/rosbag2.git (push)`. `git branch` gives master. I'm really not sure what is giving me this strange error. There is no clean command right?

Kaju-Bubanja, 2021-02-15:
So for lack of options I deleted the folders build/install/log and rebuild the workspace. But I still get the same error. The odd thing on another system I can build rosbag2_storage successfully. What other folders do I need to delete to get a clean slate? I thought with the above commands of git I verified that I really have the latest and correct package and with deleting the 3 folders I got a clean slate? I'm not sure what else it could be. There are some warnings in the build process, but I'm pretty sure I had those before and it was not a problem:
```

-- stderr: foonathan_memory_vendor                                                
Cloning into 'foo_mem-ext'...
Note: switching to 'c619113'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at c619113 Whitespace clean up (#70)
Submodule 'cmake/comp' (https://github.com/foonathan/compatibility.git) registered for path 'cmake/comp'
Cloning into '/home/usr/ros2_foxy/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/cmake/comp'...
---

```
```

-- stderr: mimick_vendor                                                    
Cloning into 'mimick-ext'...
Note: switching to '99a35f3d2067708931945c64ac9caee80a0ef50e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 99a35f3 Update cmake minimum version to 2.8.12 (#15)
---

```
```

-- stderr: rmw_connext_shared_cpp                                                      
CMake Warning at CMakeLists.txt:22 (message):
  Could not find RTI Connext - skipping 'rmw_connext_shared_cpp'

---
Finished <<< rmw_connext_shared_cpp [2.15s]
--- stderr: rosidl_typesupport_connext_cpp                                              
CMake Warning at CMakeLists.txt:32 (message):
  Could not find RTI Connext - skipping 'rosidl_typesupport_connext_cpp'

---
Finished <<< rosidl_typesupport_connext_cpp [2.31s]
Starting >>> rosidl_typesupport_connext_c
--- stderr: rosidl_typesupport_connext_c                                                
CMake Warning at CMakeLists.txt:37 (message):
  Could not find RTI Connext - skipping 'rosidl_typesupport_connext_c'

---
Finished <<< rosidl_typesupport_connext_c [1.47s]
Starting >>> rmw_connext_cpp
--- stderr: rmw_connext_cpp                                                             
CMake Warning at CMakeLists.txt:32 (message):
  Could not find RTI Connext - skipping 'rmw_connext_cpp'

---
Finished <<< rmw_connext_cpp [1.36s]
Finished <<< urdf [7.22s]                                                                
Starting >>> kdl_parser
Finished <<< kdl_parser [2.61s]                                                          
Finished <<< rcl_yaml_param_parser [16.5s]                                               
[Processing: fastrtps, qt_gui_cpp, rviz_ogre_vendor]                                     
--- stderr: qt_gui_cpp                                                                   
qt.shiboken: (typesystem) Unable to locate Clang's built-in include directory (neither by checking the environment variables LLVM_INSTALL_DIR, CLANG_INSTALL_DIR  nor running llvm-config). This may lead to parse errors.
(typesystem) clang_parseTranslationUnit2(0x0, cmd[25]=-fPIC -Wno-constant-logical-operand -std=c++14 -I/home/usr/ros2_foxy/src/ros-visualization/qt_gui_core/qt_gui_cpp/include -I/home/usr/ros2_foxy/install/pluginlib/include -I/home/usr/ros2_foxy/install/class_loader/include -I/home/usr/ros2_foxy/install/rcpputils/include -I/home/usr/ros2_foxy/install/rcutils/include -I/home/usr/ros2_foxy/install/console_bridge_vendor/include -I/home/usr/ros2_foxy/install/ament_index_cpp/include -I/usr/include -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/usr/include/PySide2 -I/usr/include/c++/9 -I/usr/include/x86_64-linux-gnu/c++/9 -I/usr/include/c++/9/backward -I/usr/lib/gcc/x86_64-linux-gnu/9/include -I/usr/local/include -I/usr/include/x86_64-linux-gnu -I/usr/include /tmp/global_yLeWgt.hpp)
qt.shiboken: (typesystem) Enum 'RCUTILS_LOG_SEVERITY' does not have a type entry
qt.shiboken: (typesystem) While traversing QOpenGLShaderProgram: setAttributeArray(int,const float*,int,int): Cannot use parameter 2 as an array: <array> modification already applied.
qt.shiboken: (typesystem) class 'QAccessibleWidget' inherits from unknown base class 'QAccessibleActionInterface'
qt.shiboken: (typesystem) class not found for setup inheritance 'QAccessibleActionInterface'
---

```

According to https://answers.ros.org/question/359355/questions-on-ros2-foxy-source-code/ I can ignore the RTI warnings. The detached head warnings also seem harmless. 

I'm not sure about the qt_gui_cpp warning though, I'm not sure if I saw it before. To be sure I don't use clang to compile I checked CC and CXX and they are empty as described here: https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#alternate-compilers `cc -v` gives gcc version 9.3.0.

Kaju-Bubanja, 2021-02-15:
Ah I found it, I pulled master of rosbag2 but not of the other packages. Once I did this as described here: https://index.ros.org/doc/ros2/Installation/Maintaining-a-Source-Checkout/#maintainingsource the error went away.

closed 2021-02-15 by Kaju-Bubanja