ros2/rosbag2 #652: Replay "zero-copy" messages

albtam, 2021-02-12:
## Description

When using zero-copy, the publisher should first borrow a loan, then populate it, and finally return the reference to the middleware. This interface is currently not available in rosbag2.

### Expected Behavior

It should be possible to publish "zero-copy" messages

### Actual Behavior

The messages stored in the rosbag are not published and the following DDS error is thrown: 

"Something went wrong when using sample state. This should never go wrong."

## To Reproduce

1. Record "zero-copy" messages in a bag
2. Replay the bag

## System (please complete the following information)
 - OS: Ubuntu 18.04.1
 - ROS 2 Distro: Fork from Dashing 
 - rosbag2 commit: a584abe4c6ebc25927458301a3d1bb57b840a0bf

## Additional context

I already implemented a fix in our fork if you are interested

emersonknapp, 2021-02-12:
Is rosbag2 able to successfully record zero-copy messages, though?

Zero copy actually just came up in the working group meeting this morning - I'm not personally that familiar with the feature, but it makes sense that rosbag2 should take advantage of it however possible to help with performance.

If you have a working implementation, a PR against `master` would be very welcome - then we could backport to any other live ros2 distros as necessary, assuming it doesn't break the API/ABI.

Karsten1987, 2021-02-12:
Happy to review incoming PRs to that topic.

However, I am having trouble following your bug description.
First off, ROS2 started supporting zero-copy messages, that is the concept of `LoanedMessages` starting from Foxy. So I wouldn't expect it to work in `Dashing` as you've mentioned. 
Then further, I don't know what zero-copy means here in terms of replaying a rosbag. You still need to copy the data from the bagfile into a loaned message in that case anyway before you can send them. Given that within the bag the data is only available as a serialized data stream, "loaning a message" from the middleware in this case really only means getting a large enough byte-buffer from the middleware without any further type information. That on the flipside entails that the subscription side should also be able to receive the data as that serialized byte buffer. ROS2 as of now doesn't have the support of loaning serialized messages in their RMW layer, so I feel that problem is a bit bigger than just rosbag2.

Nevertheless, can you elaborate on your reproducer example? I am having a hard time understanding how I can follow your steps.

albtam, 2021-02-12:
Sure, I'll provide more detail. Just to be clear @Karsten1987 , this is what we discussed at the end of last year (with Apex.AI)

albtam, 2021-02-19:
I updated the issue description. Can you please update the tag?

> First off, ROS2 started supporting zero-copy messages, that is the concept of LoanedMessages starting from Foxy. So I wouldn't expect it to work in Dashing as you've mentioned.

Our fork is based on Dashing but we backported the ROS 2 Foxy zero-copy APIs. To be more precise, we are by-passing some of the `*take_loaned_message*` APIs.

> I don't know what zero-copy means here in terms of replaying a rosbag. You still need to copy the data from the bagfile into a loaned message in that case anyway before you can send them

By zero-copy, I basically mean the zero-copy transport (which is in this case may not avoid all the copies). Even if on the rosbag2 side some copies might be needed, on the subscriber side, no copy should be required.

> Is rosbag2 able to successfully record zero-copy messages, though?

I haven't tested the "zero-copy" recording for DDS implementations that do provide a `take_serialized` method, but I think that rosbag2 does not support this use-case since as Karsten mentioned "ROS2 as of now doesn't have the support of loaning serialized messages".

MichaelOrlov, 2021-04-09:
@albtam We need to wrap up with this issue before May 1. \
After May 1 it will be API and new feature freeze for Galactic release.

@Karsten1987  Zero copy in the context of this feature means zero-copy on transport layer. \
Even though we still will need to copy messages and serialize/deserialize them during recording and playback on Rosbag2 side it could give some advantages in total possible data throughput when working locally on embedded devices. 

The real use case could be in recording data from LIDARs and/or cameras on embedded platform locally using IPC without involving transport layer with loopback.

albtam, 2021-04-24:
Closing the issue because of https://github.com/ros2/rosbag2/pull/664#issuecomment-819020914

closed 2021-04-24 by albtam