ros2/rosbag2 #655: Can not play Topics of custom messages like imgmsg(CvBridge ) and  pcl2(sensor_msgs_py) using 'ros2 bag play'

jaan1729, 2021-02-15:
## Description

I built ros2 rolling from source to use the new packages like rosbag2_py and sensor_messages_py.

I referred [kitti2bag](https://github.com/tomas789/kitti2bag/blob/master/kitti2bag/kitti2bag.py) repo and rosbag2 [test example](https://github.com/ros2/rosbag2/blob/master/rosbag2_py/test/test_sequential_writer.py) to record rosbags for Kitti dataset. 
Here is the core program I used.

```python
import os
import sys
import rclpy as rospy
from rclpy.serialization import deserialize_message, serialize_message
from rosidl_runtime_py.utilities import get_message
from std_msgs.msg import String
import transformations

if os.environ.get('ROSBAG2_PY_TEST_WITH_RTLD_GLOBAL', None) is not None:
    # This is needed on Linux when compiling with clang/libc++.
    # TL;DR This makes class_loader work when using a python extension compiled with libc++.
    #
    # For the fun RTTI ABI details, see https://whatofhow.wordpress.com/2015/03/17/odr-rtti-dso/.
    sys.setdlopenflags(os.RTLD_GLOBAL | os.RTLD_LAZY)

from .common import get_rosbag_options  # noqa
import rosbag2_py  # noqa

import tf2_ros as tf
import os
import cv2

import rosbag2_py
import progressbar
from tf2_msgs.msg import TFMessage
from datetime import datetime
from std_msgs.msg import Header
from sensor_msgs.msg import CameraInfo, Imu, PointField, NavSatFix
import sensor_msgs_py.point_cloud2 as pcl2
from geometry_msgs.msg import TransformStamped, TwistStamped, Transform
from cv_bridge import CvBridge
import numpy as np
import argparse
import pykitti

def create_topic(writer, topic_name, topic_type, serialization_format='cdr'):
    """
    Create a new topic.
    :param writer: writer instance
    :param topic_name:
    :param topic_type:
    :param serialization_format:
    :return:
    """
    topic_name = topic_name
    topic = rosbag2_py.TopicMetadata(name=topic_name, type=topic_type,
                                     serialization_format=serialization_format)

    writer.create_topic(topic)

def save_imu_data(bag, kitti, imu_frame_id, topic):
    print("Exporting IMU")
    msg_topic = topic
    create_topic(bag, msg_topic, 'sensor_msgs/msg/Imu')
    for timestamp, oxts in zip(kitti.timestamps, kitti.oxts):
        q = transformations.quaternion_from_euler(oxts.packet.roll, oxts.packet.pitch, oxts.packet.yaw)
        
        imu = Imu()
        imu.header.frame_id = imu_frame_id
        
        #imu.header.stamp = rospy.Time.from_sec(float(timestamp.strftime("%s.%f")))
        Timer = rospy.time.Time(seconds = int(timestamp.strftime("%s")), nanoseconds = int(timestamp.strftime("%f"))*10e7)
[... 364 lines cut ...]
I have successfully recorded the data. 
But when I tried publishing the bags using the following commands.

```bash
source ~/ros2_rolling/install/setup.bash && source ~/ros_pkgs/install/setup.bash
ros2 bag play temp/tmp_write_test
```

It is publishing normal topics of std_msg. But is ignoring a few topics by showing the following warning.

```bash
[INFO] [1613380015.893647871] [rosbag2_storage]: Opened database 'temp/tmp_write_test/tmp_write_test_0.db3' for READ_ONLY.
[WARN] [1613380015.899220038] [rosbag2_transport]: Ignoring a topic '/kitti/camera_gray_right/image_raw', reason: dlopen error: /home/ubuntu/ros_pkgs/install/cv_bridge/lib/libcv_bridge__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory, at /home/ubuntu/ros2_rolling/src/ros2/rcutils/src/shared_library.c:99.
[WARN] [1613380015.903106174] [rosbag2_transport]: Ignoring a topic '/kitti/camera_gray_left/image_raw', reason: dlopen error: /home/ubuntu/ros_pkgs/install/cv_bridge/lib/libcv_bridge__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory, at /home/ubuntu/ros2_rolling/src/ros2/rcutils/src/shared_library.c:99.
[WARN] [1613380015.907897351] [rosbag2_transport]: Ignoring a topic '/kitti/velo/pointcloud', reason: dlopen error: /home/ubuntu/ros2_rolling/install/sensor_msgs_py/lib/libsensor_msgs_py__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory, at /home/ubuntu/ros2_rolling/src/ros2/rcutils/src/shared_library.c:99.
[WARN] [1613380015.910327644] [rosbag2_transport]: Ignoring a topic '/kitti/camera_color_left/image_raw', reason: dlopen error: /home/ubuntu/ros_pkgs/install/cv_bridge/lib/libcv_bridge__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory, at /home/ubuntu/ros2_rolling/src/ros2/rcutils/src/shared_library.c:99.
[WARN] [1613380015.912786349] [rosbag2_transport]: Ignoring a topic '/kitti/camera_color_right/image_raw', reason: dlopen error: /home/ubuntu/ros_pkgs/install/cv_bridge/lib/libcv_bridge__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory, at /home/ubuntu/ros2_rolling/src/ros2/rcutils/src/shared_library.c:99.
```

I have cv_bridge package in ~/ros_pkgs workspace. And ros2 is built in ~/ros2_rolling and sourced both of them while building this package. And I pulled latest source and tested on it just before writing this issue.

## System (please complete the following information)
 - OS:  Ubuntu Focal
 - ROS 2 Distro: Rolling
 - Version: master

emersonknapp, 2021-02-15:
I'm just going to note that this repro code is _too long to read_ - are you able to provide a more minimal reproduction case? E.g. `ros2 topic pub cv_msgs/Whatever` piped into a rosbag2?

To address the warnings - do the mentioned files such as `/home/ubuntu/ros_pkgs/install/cv_bridge/lib/libcv_bridge__rosidl_typesupport_cpp.so` actually exist? If they are not there, are they somewhere else? (`find /home/ubuntu -name '*libcv_bridge*'`)

jaan1729, 2021-02-17:
Thanks for the responce. I tried to publish the fake data and compared with this code.

I used wrong message types while creating topics. Correcting them helped.

Ex:
1.
```python
create_topic(bag, img_topic, 'cv_bridge/CVBridge/imgmsg')  #Incorrect
create_topic(bag, img_topic, 'sensor_msgs/msg/Image')  #Correct
```

2.
```python
create_topic(bag, msg_topic, 'sensor_msgs_py/point_cloud2/pcl2')  #Incorrect
create_topic(bag, msg_topic, 'sensor_msgs/msg/PointCloud2')   #Correct
```

The issue is resolved.

closed 2021-02-17 by emersonknapp

fabrizioschiano, 2022-02-22:
Hi @jaan1729 , did you understand why `sensor_msgs_py` did not work? I m stumbling across the same problem. Thanks!