ros2/rosbag2 #663: Support rosbag snapshot

d-walsh, 2021-02-18:
## Description

The snapshot capability was added to the ROS1 version of rosbag and it would be great to support this use case with ROS2. https://github.com/ros/rosbag_snapshot

emersonknapp, 2021-02-18:
Really cool feature :+1:

emersonknapp, 2021-02-26:
Would like to have more details of the feature on this ticket itself. Putting to bottom of backlog for now.

kielczykowski, 2021-04-14:
Hey, it feels like i can give a hand with implementing this feature.

Would it be more suitable to open PR and new discussion there or continue designing solution in this thread?

emersonknapp, 2021-04-14:
As with most features of any significant weight, I'd prefer a design proposal first before looking at an implementation (feels slower - but slow is smooth and smooth is fast). You could open up a PR with a new document in the folder https://github.com/ros2/rosbag2/tree/master/docs/design - then we could tie this feature request ticket to that design.

There's room for flexibility - if you think it's faster to throw up a sample implementation than to write a doc - but generally I find it easier to review the doc, that way any disagreement over definitions or behaviors can be hashed out before a lot of time is lost on making it work well enough for a proof-of-concept.

Luckily, we're getting to a place where it should be a lot easier to implement something like this. The `Recorder` has been converted to be a `rclcpp::Node` which will make it easier to expose new APIs via services (I am currently implementing the same idea for playback control on `Player`)

In a design, I would be looking for specifics on expected user inputs / disk outputs. It doesn't have to focus too heavily on the exact implementation details, but pseudocode API or diagrams with an idea of how they tie to existing structures would help make the review go smoothly. Feel free to join the [Tooling WG](https://docs.google.com/document/d/1UXWaEkT6ji0Z0uTtJJSni5Y6fiopM9qOOzv4ldeNaAM) meetings if you want to have open discussion.

closed 2021-09-25 by emersonknapp

JWhitleyWork, 2021-10-21:
@emersonknapp I am apparently blind but thanks for pointing this out! Is there documentation for usage?

emersonknapp, 2021-10-21:
You know what - that made me realize there isn't! I'll look into adding some now.

You can see the option in `ros2 bag record --help` - but we should at least have a mention in the README.

A more comprehensive documentation page is definitely needed but that's a little ways out, I think.

JWhitleyWork, 2021-10-21:
Thanks! I'll give it a look and see how it compares.

PR #851 (2021-09-25): Add --snapshot-mode argument to the "record" verb
Closes #663
Extends #844
Extends #850

This PR adds the `--snapshot-mode` argument to the Python CLI `record` verb. It also updates the original design document to reflect minor changes to the implementation.