ros2/rosbag2 #665: How to set up launch arguments for multiple topics?

khaiyichin, 2021-02-23:
## Description
This is more of a question, but I couldn't seem to find any pointers on how to do it. Essentially I'm trying to create a launch file that allows the user to specify the topics as arguments to the launch file. I wasn't sure if this was a `launch` question, but I figured it might be helpful to have a `rosbag2` targeted answer.

## To Reproduce
Minimum example:
```python3
# rosbag_record.launch.py
launch_arg_rosbag_topics = DeclareLaunchArgument('topics', default_value='-a')

process_rosbag_record = ExecuteProcess(
      cmd=['ros2', 'bag', 'record', LaunchConfiguration('topics')],
)

return LaunchDescription([
    process_rosbag_record
])
```
In the terminal,
```bash
# terminal
$ ros2 launch ./rosbag_record.launch.py topics:='/cmd_vel /odom /odom_gt /velodyne_points /tf /tf_static'
```
gives the following error:
```bash
[ros2-1] [ERROR] [1614037748.512620668] [rosbag2_transport]: Failed to expand topic name /cmd_vel /odom /odom_gt /velodyne_points /tf /tf_static with error: topic name is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.10/src/rcl/expand_topic_name.c:73
```
And after killing it I get the following message, which seems to indicate that the evaluated command should be correct, and I was able to verify by copy and pasting it (i.e., `ros2 bag record /cmd_vel /odom /odom_gt /velodyne_points /tf /tf_static`).
```bash
[ERROR] [ros2-1]: process has died [pid 257759, exit code 2, cmd 'ros2 bag record /cmd_vel /odom /odom_gt /velodyne_points /tf /tf_static'].
```
## System (please complete the following information)
 - OS: Ubuntu Focal
 - ROS 2 Distro: Foxy

emersonknapp, 2021-02-23:
This question may be better suited to answers.ros.org

emersonknapp, 2021-03-05:
To add some more context on this - the commandline is being interpreted as a request to record a _single_ topic named `"/cmd_vel /odom /odom_gt /velodyne_points /tf /tf_static"` - which is not a valid topic name because it contains spaces. The ExecuteProcess args need to be a list. This is described in https://github.com/ros2/launch/issues/263 

Until that issue is solved, you can solve this using `shell=True` as is described in the above issue. E.g.

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import ExecuteProcess
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('topics', default_value=['-a'], description='Topics to record'),
        ExecuteProcess(cmd=['ros2', 'bag', 'record', LaunchConfiguration('topics')], shell=True),
    ])
```

I'll close this issue given there is nothing to do in rosbag2 for this.

closed 2021-03-05 by emersonknapp