ros2/rosbag2 #672: Record topic before existence of the topic

Patrick-Lascombe, 2021-03-08:
## Description

By going through the code of rosbag2, I stumbled into this comment in _rosbag2_transport.hpp_ : `
Records topics to a bagfile. Subscription happens at startup time, hence the topics must exist when "record" is called.` and I wanted to know why this choice was made in ROS2 when the feature to record yet non-existant topic in ROS1 exist ?

I have not found an issue or PR referring this features and it looks link an important enhancement to me.

emersonknapp, 2021-03-08:
The comment is (partially) out of date - a discovery loop is run until all requested topics are found. However, in the case of `-a` specifically, subscription happens on all topics that are discovered at startup time, then discovery is stopped. So if you are requesting to record _specific_ topics, it is ok if they do not exist at the start.

emersonknapp, 2021-03-26:
I'm going to close this as stale for now - please reopen it if the above comment didn't resolve your concern.

closed 2021-03-26 by emersonknapp