ros2/rosbag2 #682: Support record enhancements from rosbag_fancy - UI and Throttling

d-walsh, 2021-03-17:
## Description

rosbag_fancy, created by @xqms, is a great tool for recording rosbags in ROS1: https://github.com/xqms/rosbag_fancy

It adds two really useful features:
1. A live UI for monitoring the recording with useful metrics
2. The ability to throttle the rate of recording per topic

It would be great if these features could be integrated into rosbag2 record.

Here's the UI:
[image]

emersonknapp, 2021-03-17:
This is a cool feature set - I would love to see a design proposal for it before an implementation, if anybody decides to pick this up.

xqms, 2021-03-18:
Hey, author of `rosbag_fancy´ here. I wouldn't want to lead development of a rosbag2 port, since I don't use ROS 2 myself (yet),
but I'm available for design discussion / help :)