ros2/rosbag2 #685: can we run rosbag2 record / play as a ROS2 node or component?

hakuturu583, 2021-03-19:
## Description

Hi, I an developing ROS2 application and I found rosbag2 packages has no executable and components to play or record topics.

## Implementation Notes / Suggestions

adding player and recorder component by using rosbag2_cpp::Reader and rosbag2_cpp::Write

emersonknapp, 2021-03-19:
Good call! It makes perfect sense to provide a node/component that can be used, rather than having to call it via the CLI (or write your own program). This should not be hard to implement, the trickiest part will be making it configurable by parameters instead of commandline options.

Implementation notes: create a `transport_node` component in `rosbag2_transport` library. Use parameters to configure its options. Setup of the record/playback itself will look similar to https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/rosbag2_transport_python.cpp#L224

hakuturu583, 2021-03-22:
@emersonknapp Thank you for your advice!

fujitatomoya, 2021-03-22:
I guess that we tend to support this feature as general class, that is why @emersonknapp put `help wanted` tag. (if i am not mistaken...) we would probably keep this open? 🤔

emersonknapp, 2021-04-09:
It's looking like this will probably be possible in the Galactic release - the Recorder and Player are becoming nodes, which could be components, configurable with parameters.

MichaelOrlov, 2023-03-09:
@tonynajjar No updates about this task. I am sorry, don't have resources for that for a while.
Help would be appreciated.

MichaelOrlov, 2023-09-08:
This issue will be resolved in the #1419

closed 2023-10-17 by hakuturu583