ros2/rosbag2 #694: play - drive rosbag2 playback via external /clock time source

emersonknapp, 2021-03-26:
## Description

Allow rosbag2 to listen to the `/clock` topic as its time source. This would allow for rosbag2 to play in sync with Gazebo, or to synchronize multiple bag playbacks with each other (one publishes `/clock`, the rest subscribe)

This should probably be invoked by `ros2 bag play --use-sim-time`

## Related Issues

Related to #99 
Designed by #675 

## Implementation Notes / Suggestions

See #675 

## Testing Notes / Suggestions

Run a bag playback with a separate clock publisher, and see that messages are published when expected.

emersonknapp, 2023-07-19:
Closing as completed

giovannirizzetto, 2023-08-03:
Hello, 

this rosbag2 feature would be very important in a project I am working on. I have not understood if it is going to be available or not, seems to me it was abandoned, right?

Thanks a lot for your assistance

emersonknapp, 2023-08-03:
Hmmm - I remember being quite sure 2 weeks ago that this had been completed somehow - but I'm looking at the code now and not seeing any evidence of this. Maybe I was distracted at the time?

emersonknapp, 2023-08-03:
I don't think this would be too hard to implement. https://github.com/ros2/rosbag2/pull/928/files was definitely close to it. I'm not sure now why we closed it.

emersonknapp, 2023-08-20:
Well, unfortunately I don't think anybody has bandwidth allotted for this feature at the moment. If you wanted to grab the branch from #928 and take over finishing the implementation that would be great, and I'd be happy to try and give it a timely review.