ros2/rosbag2 #696: [Epic] Rosbag2 playback time control and simulation time

emersonknapp, 2021-03-26:
## Description

Tracking issue for the group of features (planned for Galactic) around control of the flow of time during rosbag2 playback. See "Completion Criteria" for full feature list

Design: #675 

## Completion Criteria

Design at https://github.com/ros2/rosbag2/blob/master/docs/design/rosbag2_playback_time.md is fully implemented including all of the following features:

* [ ] pause/resume time during playback
* [ ] change rate during playback
* [ ] pause/resume/rate controllable by keyboard handler (also rqt_bag)
* [ ] jump/seek to arbitrary time during playback
* [ ] jump/seek controllable by 
* [ ] can publish to `/clock` during playback
* [ ] can subscribe to `/clock` during playback
* [ ] can extrapolate time for high frequency topics when subscribing to `/clock`

## Work Log

* [x] Design: https://github.com/ros2/rosbag2/pull/675
* [ ] Fix test bug blocking other features: https://github.com/ros2/rosbag2/pull/690
* [ ] Migrate `play` to `rosbag2_py` to make it easier to add new playback options: https://github.com/ros2/rosbag2/pull/693
* [ ] https://github.com/ros2/rosbag2/pull/689
* [ ] https://github.com/ros2/rosbag2/pull/695

Karsten1987, 2021-04-02:
Once https://github.com/ros2/rclcpp/pull/1452 is in, I'd take the refactoring of removing the `rosbag2_node` and replace it with the new rclcpp one. Does this sound ok or do I disturb other people's workflow with this? 
I feel that the task list above might lead to quite a few conflicts as they massively overlap.

emersonknapp, 2021-04-02:
I think that would be fine - I don't see it interfering very much with the existing line items. Most of the changes now should be very focused, and not be near the direct usage of `GenericPublisher`

Karsten1987, 2021-04-02:
Ok, I was just afraid that I would overlap with @MichaelOrlov's last task: "Split Rosbag2Transport into separate Player and Recorder classes"

emersonknapp, 2021-04-02:
Yes, that one could overlap, it would probably be good to stage those serially.

emersonknapp, 2021-04-02:
But - maybe it won't be too bad! Perhaps you can stage it as "replace rosbag2::generic with rclcpp::generic" - which definitely will have little to no conflict - then do "remove the rosbag2node" as a second stage

MichaelOrlov, 2021-04-03:
@Karsten1987 Yeah I think it would overlap. \
Please go ahead first. I can try to tackle another tasks for a while. Such as `jump`, `play_next` and keyboard handling.

Kaju-Bubanja, 2021-05-04:
I don't see this awesome feature in the list of new features here: https://docs.ros.org/en/foxy/Releases/Release-Galactic-Geochelone.html#timeline-before-the-release Is it still planned to be released for galactic?

emersonknapp, 2021-05-04:
Yes - everything that is currently checked off will be in Galactic (and maybe a couple more small things). None of the rosbag2 changes are listed on that page.

clalancette, 2021-05-04:
> None of the rosbag2 changes are listed on that page.

I hope you mean "yet" :smile: .  It would be awesome to have a high-level overview of the (massive) changes to rosbag2 listed on that page.

emersonknapp, 2021-05-04:
> I hope you mean "yet" smile . It would be awesome to have a high-level overview of the (massive) changes to rosbag2 listed on that page.

Yes, yet. Should we open a PR against the page source for it?

clalancette, 2021-05-04:
> Yes, yet. Should we open a PR against the page source for it?

Yes, whenever you are ready, please feel free to open a PR and I'm happy to review it.

emersonknapp, 2021-11-01:
Closing this issue, we don't need it anymore. It's a huge amount of work that we did! The only thing really remaining is use-sim-tim, which can be tracked via its own issue https://github.com/ros2/rosbag2/issues/694

closed 2021-11-01 by emersonknapp