ros2/rosbag2 #698: Wrong include directories order

hexonxons, 2021-03-28:
## Description
I've tried to build https://github.com/ros2/rosbag2/pull/687 to check if it works in our environment, but it seems it wont build clean if ros-foxy-rosbag2... packages are installed on the target machine.

I've checked it while building on host machine and reproduced in `ros:foxy` docker container. In both cases `colcon build` finishes with an error:
<details>
    <summary>ERROR</summary>

```
--- stderr: rosbag2_transport_future                                      
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:42,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_reader.hpp:28:8: error: ‘void MockSequentialReader::open(const rosbag2_storage::StorageOptions&, const rosbag2_cpp::ConverterOptions&)’ marked ‘override’, but does not override
   28 |   void open(
      |        ^~~~
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:43,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_writer.hpp:28:8: error: ‘void MockSequentialWriter::open(const rosbag2_storage::StorageOptions&, const rosbag2_cpp::ConverterOptions&)’ marked ‘override’, but does not override
   28 |   void open(
      |        ^~~~
In file included from /usr/include/c++/9/memory:80,
                 from /opt/ros/foxy/src/gmock_vendor/include/gmock/gmock-actions.h:45,
                 from /opt/ros/foxy/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:15:
/usr/include/c++/9/bits/unique_ptr.h: In instantiation of ‘typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = MockSequentialReader; _Args = {}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<MockSequentialReader, std::default_delete<MockSequentialReader> >]’:
/ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:62:90:   required from here
/usr/include/c++/9/bits/unique_ptr.h:857:30: error: invalid new-expression of abstract class type ‘MockSequentialReader’
  857 |     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:42,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_reader.hpp:25:7: note:   because the following virtual functions are pure within ‘MockSequentialReader’:
   25 | class MockSequentialReader : public rosbag2_cpp::reader_interfaces::BaseReaderInterface
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_reader.hpp:23,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:42,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/opt/ros/foxy/include/rosbag2_cpp/reader_interfaces/base_reader_interface.hpp:40:16: note:  ‘virtual void rosbag2_cpp::reader_interfaces::BaseReaderInterface::open(const rosbag2_cpp::StorageOptions&, const rosbag2_cpp::ConverterOptions&)’
   40 |   virtual void open(
      |                ^~~~
In file included from /usr/include/c++/9/memory:80,
                 from /opt/ros/foxy/src/gmock_vendor/include/gmock/gmock-actions.h:45,
                 from /opt/ros/foxy/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:15:
/usr/include/c++/9/bits/unique_ptr.h: In instantiation of ‘typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = MockSequentialWriter; _Args = {}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<MockSequentialWriter, std::default_delete<MockSequentialWriter> >]’:
/ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:63:90:   required from here
/usr/include/c++/9/bits/unique_ptr.h:857:30: error: invalid new-expression of abstract class type ‘MockSequentialWriter’
  857 |     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:43,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_writer.hpp:25:7: note:   because the following virtual functions are pure within ‘MockSequentialWriter’:
   25 | class MockSequentialWriter : public rosbag2_cpp::writer_interfaces::BaseWriterInterface
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /ws/src/rosbag2_transport/test/rosbag2_transport/mock_sequential_writer.hpp:23,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/rosbag2_transport_test_fixture.hpp:43,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:31,
                 from /ws/src/rosbag2_transport/test/rosbag2_transport/test_record_regex.cpp:24:
/opt/ros/foxy/include/rosbag2_cpp/writer_interfaces/base_writer_interface.hpp:37:16: note:  ‘virtual void rosbag2_cpp::writer_interfaces::BaseWriterInterface::open(const rosbag2_cpp::StorageOptions&, const rosbag2_cpp::ConverterOptions&)’
   37 |   virtual void open(
      |                ^~~~
make[2]: *** [CMakeFiles/test_record_regex__rmw_fastrtps_cpp.dir/build.make:63: CMakeFiles/test_record_regex__rmw_fastrtps_cpp.dir/test/rosbag2_transport/test_record_regex.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:512: CMakeFiles/test_record_regex__rmw_fastrtps_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
[... 28 lines cut ...]
Everything compiles successfully and include directories order is right

### Actual Behavior
Error while compiling rosbag2 packages

## To Reproduce
1. Run ros container with rosbag2 inside:
```
docker run --rm -it ros:foxy /bin/bash
```
2. Create workspace and clone rosbag2: 
```
mkdir -p ws/src
git clone https://github.com/ros2/rosbag2.git -b emersonknapp/futurize-foxy ws/src
cd ws
```
3. Build:
```
colcon build --ament-cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
```

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: foxy
 - Version: Reproduced in emersonknapp/futurize-foxy, but master branch still affected

emersonknapp, 2021-04-07:
> It seems that adding rosbag2_cpp_future to AMENT_DEPS for test_record_regex moves -isystem /ws/install/rosbag2_cpp_future/include to position before -isystem /opt/ros/foxy/include but I am not sure [...]

I think that this is the right fix - most of the time this type of problem happens because target aren't declaring their dependencies explicitly, so if there is a system-installed version it gets picked up by accident. PRs welcome

emersonknapp, 2021-04-09:
Note: we probably want a regular build that tries to build from source, when the binary package is already installed. We have no way to catch these problems in the current setup

emersonknapp, 2021-07-15:
This is a symptom of https://github.com/ros2/ros2/issues/1150, right?

MichaelOrlov, 2023-02-01:
- Closing as stale issue.

PR #780 (2021-06-16): Fix/futurize foxy dependency fixes
Fix #698 

Needs to be refactored after https://github.com/ros2/ros2/issues/1150

How to check if everything goes "OK":
* Build via `colcon build --ament-cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_ROSBAG2_BENCHMARKS=ON --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_ROSBAG2_BENCHMARKS=ON`
* Open `build/compile_commands.json`
* Test against regex `-((I)|(isystem))\s*/opt(?!((/ros/foxy/src/gmock)|(/ros/foxy/src/gtest))).*-((I)|(isystem))\s*/home`, where `/home` - first part of absolute path to local workspace
closed 2023-02-01 by MichaelOrlov