ros2/rosbag2 #703: Question on how to analyze data using ros2 bag [.db3](3 methods)

LEE-HONGGYU, 2021-03-31:
**## Description**

I want to visualize end-to-end data at once using the bag file (db3) of ros2. So far, I have found three ways.

1. To use various visualization codes used in ros, use SQLite.Switch db3 file to .bag file and apply
2. Visualization method using Python code using .db3 file
3. Convert the .db3 file to csv or txt and plot it => I have limit with UTF-8 convert, what should I do?
4. Is there any additional way? [-s {--storage} way did not work for me. I just want exact way]

**## Related Issues**

 #561

**## Completion Criteria**

Python code or method for visualizing a ros2 bag file

**## Implementation Notes / Suggestions**

I want to see the corresponding end-to-end data as a plot when entering the desired topic (node) in the ros2 bag file. In particular, it would be better if the x and y axis limit were also specified as the maximum minimum. In addition, I would like to see various visualizations through various plots, not just one plot.

**## Testing Notes / Suggestions**

[MANUAL01.zip](https://github.com/ros2/rosbag2/files/6236332/MANUAL01.zip)

**## SYSTEM STATUS**

- Linux : ubuntu 20.04 [Foxy]
- ROS2 [ROS1 is also available on other computers, so anything is only available.]

emersonknapp, 2021-04-09:
Note to self: This is asking for a Python interface that can read a topic from a bag and produce a 2D plot of the data.

facontidavide, 2021-05-20:
@ghdrb9138 , Have you tried to used PlotJuggler?

closed 2021-05-20 by ghdrb9138

GRINGOLOCO7, 2024-04-11:
Hi,
I find myself in the same position,
How do I convert my bag (.db3) to a .csv file to see visually the data?
@ghdrb9138 how did you sole this issue?
Thank you

nayak2358, 2024-05-24:
@facontidavide PlotJuggler works well, thank you! With a lot of recorded topics, it will be great if there is a script to read the ros2 bags and output the csv file in the same format as PlotJuggler does it.

facontidavide, 2024-05-25:
Using my other package, rox_introspection, it can be done relatively easily.

Someone should... You maybe?

nayak2358, 2024-05-25:
I don't know enough C++ yet to implement it using your package. So, I found a python API and added some lines to the proposed deserialization script here : https://github.com/ros2/rosbag2/issues/473#issuecomment-2131382174