ros2/rosbag2 #706: Unable to build rosbag2_storage

FullMetalNicky, 2021-04-01:
## Description
I have ROS2 Foxy build from source, and everything works fine there.
I created a new workspace for rosbag2, as suggested in the readme. 
I cloned foxy-future and tried to build the package with colcon build --merge-install

### Expected Behavior
I expected the packaged to be built, so I can use the python-API for rosbag2

### Actual Behavior
The build failed on rosbag2_storage (see additional context)

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. git clone rosbag2 from foxy-future
2. colcon build --merge-install

## System (please complete the following information)
 - OS: Ubuntu 20.4
 - ROS 2 Distro: Foxy
 
## Additional context
Terminal output:
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_test_common
Starting >>> zstd_vendor
Finished <<< sqlite3_vendor [0.10s]                                   
Finished <<< rosbag2_test_common [0.11s]
Starting >>> rosbag2_storage
Finished <<< zstd_vendor [0.12s]
Finished <<< shared_queues_vendor [0.13s]
--- stderr: rosbag2_storage                              
/home/nickybones/Code/rosbag2_ws/src/rosbag2/rosbag2_storage/src/rosbag2_storage/metadata_io.cpp: In member function ‘virtual rosbag2_storage::BagMetadata rosbag2_storage::MetadataIo::read_metadata(const string&)’:
/home/nickybones/Code/rosbag2_ws/src/rosbag2/rosbag2_storage/src/rosbag2_storage/metadata_io.cpp:236:53: error: could not convert ‘& metadata.rosbag2_storage::BagMetadata::bag_size’ from ‘uint64_t*’ {aka ‘long unsigned int*’} to ‘rcutils_allocator_t’
  236 |       rcutils_calculate_directory_size(uri.c_str(), &metadata.bag_size, allocator))
      |                                                     ^~~~~~~~~~~~~~~~~~
      |                                                     |
      |                                                     uint64_t* {aka long unsigned int*}
make[2]: *** [CMakeFiles/rosbag2_storage.dir/build.make:63: CMakeFiles/rosbag2_storage.dir/src/rosbag2_storage/metadata_io.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:129: CMakeFiles/rosbag2_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_storage [0.53s, exited with code 2]

Summary: 4 packages finished [0.77s]
  1 package failed: rosbag2_storage
  1 package had stderr output: rosbag2_storage
  10 packages not processed

emersonknapp, 2021-04-01:
`master` is the Rolling / Galactic source branch - it is not expected to build against a Foxy installation. Additionally - `foxy-future` is not yet implemented, if you would like to try building it now, please check out the branch from the PR https://github.com/ros2/rosbag2/pull/687

So - this behavior is currently as expected - I will close this ticket.

closed 2021-04-01 by emersonknapp

narutojxl, 2022-03-10:
My OS is ubuntu20.04, installed foxy by debian. I want to play ros2 bag offline to test a slam and localization system. `/clock` topic is not published by anyone. I see in the https://github.com/ros2/rosbag2/issues/99#issuecomment-904837893 says,  "It is available in Galactic and Rolling distributions". But in `master` and `galactic`, can not build successfully, as the same error, 
> rosbag2_storage/metadata_io.cpp:236:53: error: could not convert ‘& metadata.rosbag2_storage::BagMetadata::bag_size’ from ‘uint64_t*’ {aka ‘long unsigned int*’} to ‘rcutils_allocator_t’

Git checkout `foxy-future`, build with `colcon build --symlink-install`, `rosbag2_transport` failed.  
>
	Summary: 11 packages finished [35.2s]
	  1 package failed: rosbag2_py
	  2 packages aborted: rosbag2_compression_zstd rosbag2_transport
	  2 packages had stderr output: plotjuggler rosbag2_py
	  4 packages not processed
After source workspace, `ros2 bag info` and `ros2 bag play` can not work.
>
	ros2 bag info rosbag2_2022_02_28-02_24_37_0.db3  
	Traceback (most recent call last):
	  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
	    load_entry_point('ros2cli==0.9.10', 'console_scripts', 'ros2')()
	  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
	    rc = extension.main(parser=parser, args=args)
	  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/command/bag.py", line 38, in main
	    return extension.main(args=args)
	  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2bag/verb/info.py", line 40, in main
	    from rosbag2_transport import rosbag2_transport_py
	  File "/opt/ros/foxy/lib/python3.8/site-packages/rosbag2_transport/__init__.py", line 17, in <module>
	    rosbag2_transport_py = import_c_library('._rosbag2_transport_py', package='rosbag2_transport')
	  File "/opt/ros/foxy/lib/python3.8/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
	    return importlib.import_module(name, package=package)
	  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
	    return _bootstrap._gcd_import(name[level:], package, level)
	ImportError: /opt/ros/foxy/lib/librosbag2_transport.so: undefined symbol: _ZN11rosbag2_cpp6Writer4openERKNS_14StorageOptionsERKNS_16ConverterOptionsE
	The C extension '/opt/ros/foxy/lib/python3.8/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-38-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions