ros2/rosbag2 #717: ROS 2 Bag Converter

jhdcs, 2021-04-06:
## Description

The ROS 2 Bag converter would be a program/function that you could run on your current ROS 2 bags, and convert them into different storage types. This could be a different version of the same storage (for instance, upgrading formats within SQLite databases), or converting to a completely new storage type (for instance, ROS 1 bag to SQLite database).

While the spawning comment referred to the converter being used to upgrade bag formats, I feel that it should be possible to implement this generically enough that you could convert from any supported bag type to any other supported bag type. In theory, at any rate.

## Related Issues

Spawned from a conversation in issue #677 

## Completion Criteria

In order to be a _true_ bag converter, it must do the following:

- [ ] Read/parse from any valid ROS 2 bag format, _both old and new_
- [ ] Convert parsed data into a neutral intermediary format
- [ ] Write from neutral intermediary format  to any valid ROS 2 bag format, _both old and new_
- [ ] Report any data that would be lost due to the transference to the new format
- [ ] Report any data that would be missing from the new format

## Implementation Notes / Suggestions

The reason I suggest parsing to a neutral intermediary format is that it may make writing plugins for this converter a little easier. Instead of a plugin developer having to worry about transforming data between their format and any other format, all they would need to worry about is transforming to/from a known intermediary format. This may also make it easier for us to keep track of what things each bag format provides/needs, and check for things not "matching up".

An issue I foresee with this approach, however, is we will still need a reader and writer for each and every bag format that we want to write to/from. If we start creating lots of bag formats/versions, this might become difficult to maintain. We may be able to offload some of the maintenance responsibilities by allowing the converter to work with plugins - people could write plugins for new formats and versions, and as long as the plugins can read/write to/from the neutral intermediary format, everything should "just work".

## Testing Notes / Suggestions

For testing, we will need a sample, generic bagfile that contains at least a little data. We would need to test the ability of the converter to read this bagfile into the neutral intermediary format. Additionally, we would need to run a test where we take a valid neutral intermediary format, and write it to a bagfile. Ideally, we will have multiple neutral intermediary format tests: Some will contain insufficient data for the new format (thus checking to make sure the writer can produce acceptable defaults), and some providing too much data for the new format (that the writer should ignore, but report to the user that not all data will be converted).

emersonknapp, 2021-04-06:
Thanks for opening this ticket! We've been having this exact conversation on https://github.com/ros2/rosbag2/pull/709

emersonknapp, 2021-04-06:
I'd like to figure out how to make a differentiation between

1. "upgrade an old bag format to the current bag format" - this prompts a larger discussion about how we handle bag format versioning (metadata format or storage format)
2. "rewrite a bag within the same version, with different configuration" - which is what we're discussing on #709. I think there exists a trivial implementation for this.

jhdcs, 2021-04-06:
> I'd like to figure out how to make a differentiation between
>
>     1. "upgrade an old bag format to the current bag format" - this prompts a larger discussion about how we handle bag format [...]

That is a good question, that I'm not sure I fully addressed/touched on in my original post. I'll try to expand my thoughts on the matter here.

## Definitions

First, a couple definitions so that I don't confuse anyone (like myself) with what I'm talking about. We have `Storage Types`, which indicate what the underlying storage mechanism of the bag is/will be. This is like SQLite, ROS1 bags, etc. Within each `Storage Type`, we would have a `Storage Format`. This is HOW the data is stored within that type - i.e: What data gets put in, and in what order. A `Storage Version` is some sort of identifier that we apply to a `Storage Format` to keep track of what we expect should be there.

In my mind, the metadata format/version and storage/version are **NOT** equivalent. In future, we may add in a parameter to the metadata to keep track of what storage format/version is in use by the bag, but I feel that the underlying storage format should be allowed to evolve/update as more and better techniques are developed to store the data, so long as they do not necessitate needing to significantly modify the metadata file (a significant modification would be the addition/removal of key-value pairs from the metadata file)

With those definitions/thoughts out of the way, I'll apply it to the two issues you've raised:

## 1. Upgrade an old bag format to the current bag format

I'm not sure that the converter would necessarily be intelligent to know what the "current" format of a bag is. It would merely understand that you wanted to convert from one `Storage Type`-`Storage Format` pair to another. It would just happen in this case that the `Storage Type` is unchanged, and the data within is just being laid out slightly differently. This was the use case that I figured you were talking about here (https://github.com/ros2/rosbag2/issues/677#issuecomment-801468839), and what spawned this issue initially. I just realized that once you're transforming data into a different format, it's not a _gigantic_ logical leap to transform it into a different `Storage Type`. You're still just "transforming data", it's just that your transformation has gotten a bit bigger.

## 2. Rewrite a bag within the same version, with different configuration

I am not sure _exactly_ what is meant by this, but I'm guessing from the issue that you linked that you are referring to the "stitcher" that is under development. I don't think the converter would really do this. In my mind, it's supposed to be more focused on transforming between `Storage Type`-`Storage Format` pairs. It may _call_ the "stitcher" as part of the transformation process, but that wouldn't be it's primary focus.

Though, if people disagree with me, and think that they _do_ in fact cover the same ground, I don't think that extending the API to just route a call to the "stitcher" would be a bad thing. If an end user calls the converter and asks it to stitch a file, they don't want a lecture from me about how they should _actually_ be calling the "stitcher". They want their bag stitched!

emersonknapp, 2021-04-06:
> you are referring to the "stitcher" that is under development

If you read through the discussion, you'll see that "Stitcher" was an inappropriate name for the API that will be presented there, and we are discussing the proper naming for it. I've suggested "Rewriter" - which starts to look a lot like a "Converter", which is why it is good for us to be very precise with definitions of terms.

> Definitions

Let me propose a slightly different definition structure according to the implementation:

* `Storage Implementation` - provided by a specific plugin, e.g. `sqlite3`, `rosbag2_bag_v2`, `leveldb`
* `Storage Implementation Version` - this is not something that has been implemented to date - this would be for example the sqlite database schema, if it were to change between rosbag2 versions
* `Bag Metadata Version` - the format version of the `metadata.yaml` file. This has evolved a few times and currently has simple if/then parsing for older versions

What you have called a "Type/Format Pair" looks to me like the combination of
* `rosbag2_storage::StorageOptions`
* `rosbag2_transport::RecordOptions`

I'd like to call this `StorageOptions+RecordOptions` combo a `Storage Configuration` - in combination they define everything that there is to know about a recorded bag, given by the fact that the `record` API takes these as input.

Side note: These two structures may not actually be currently split along the right boundaries. They also happen to contain some "runtime implementation details" (such as number of compression threads, message queue size), which we can ignore for the sake of this discussion.  This side note suggests to me that we might want to reconsider the struct groupings.

Now I think the question becomes - what's the functionality we want within that definitional framework? I think it is:

"Take any existing bag, and write it to a new bag with a new `StorageOptions+RecordOptions` configuration"

The way I'm seeing it, this functionality is as flexible as it gets, and assuming the `Reader` is capable of reading the input bag, then the implementation looks like:
* Create `Reader(input_bag)`
* Create `Writer(output_bag, output_configuration)`
* `while reader.has_next(): writer.write(reader.next())`

jhdcs, 2021-04-06:
> If you read through the discussion, you'll see that "Stitcher" was an inappropriate name for the API that will be presented there, and we are discussing the proper naming for it. I've suggested [...]

I just found the portion you were talking about, thanks for pointing it out! I had missed that. And I absolutely agree - it's best to have precise definitions so that everyone's in sync before starting on a potentially large project like this.

> Let me propose a slightly different definition structure according to the implementation:

I think your definition is much better than mine. Let's go with yours moving forward.

> What you have called a "Type/Format Pair" looks to me like the combination of
> 
>     * `rosbag2_storage::StorageOptions`
> 
>     * `rosbag2_transport::RecordOptions`

Ah! Excellent! That will simplify things a great deal, I think.

> Now I think the question becomes - what's the functionality we want within that definitional framework?

I think that the functionality that you describe in this section is pretty much what we should go for, however I would like to add that the converter should be able to warn the user if a 1:1 conversion is not possible. For instance, if the existing bag has keys stored that the new bag can't store, the converter should warn that some keys will be lost, and then list them. Similarly, if the new bag has keys that the existing bag didn't have, it should create the bag with some sort of default values for those new keys, but then warn the user that it had to create defaults for those keys.

This is partially why I was thinking of a "neutral intermediary format". It would allow you to load up the data and quickly check for missing/added keys like I outlined in the previous paragraph, but at the added cost of extra time and possibly RAM usage. An alternative could also be to somehow parse the bag's first data row, and try to extract what keys are different between the input/output configurations. This would rely on a row-based storage solution, though.

Or, whoever implements each reader and each writer has some sort of hardcoded keys correspond with each schema, and their defaults. When matching them up, the converter will be able to compare these two sets of keys for differences. It would also have the added bonus that the reader would be able to check for corrupted input bags... Potentially...

> The "old bag" question was related to the `Storage Implementation Version` - or, is the storage implementation allowed to update its schema over time, and if so how does it read those older schemas?

I think that the storage implementation should be allowed to update its schema over time. Each new schema would be a new `Storage Implementation Version` with its own reader and writer. Ideally, this shouldn't have to happen too often, but this way we can support the most conversions, and have more breathing room if we need to change schemas (since we can tell people to update the schema, and point them to this tool to do it for them).

> I was referring to the invoked rosbag2's current implementation version for `Bag Metadata Version` and `Storage Implementation Version` - by "current format" i meant "the thing that happens by default when you write a bag with rosbag2", or "the version of metadata and storage that is implemented by my installed version of rosbag2"

Ah, okay. This will not be as convenient for people who do not decide to go with Rosbag2's default metadata and storage... But if they're not going with them, things aren't as convenient anyways. So this should probably be fine to include.

emersonknapp, 2021-04-06:
> This is partially why I was thinking of a "neutral intermediary format". It would allow you to load up the data and quickly check for missing/added keys like I outlined in the previous paragraph, [...]

When you say "keys" here, can you clarify what you mean? 

Are you talking about the structure of the messages themselves (the `IDL` / `.msg`  definitions)? This isn't something I had considered - it's probably a separate thing. The structure I've put forth here doesn't really take this into consideration. I think it would be part of the `SerializationFormatConverter` portion of rosbag2 functionality, and could very well be useful as a generic ros2 core feature.

Something like:
* Read a serialized message (`cdr`) and its `Type`
* Try and find/load the current definition of `Type` from the ros2 installation
* If the type isn't installed, fail?
* If the type is installed, use serialization format introspection on the input, and rmw typesupport introspection on the loaded type, compare, and translate as many fields as possible.

That may be a big enough feature to have its own standalone ticket, like "Tanslate older CDR messages to newer message definitions". The "translate serialization format to generically introspectable key/value tree" could be implemented by specific serialization format plugins, and then a common bit of code could implemented the conversion from that to the newer message definition. <-- I see now that I've gotten to where you were with "neutral intermediary format". It wasn't clear to me that's what you'd been talking about.

jhdcs, 2021-04-07:
My apologies! When I'm talking about "keys" I'm talking about keys to the individual portions of the recorded data in the database. So the schema labels.

For example, the keys for the default SQLite storage would be what's here: https://github.com/ros2/rosbag2/blob/687f19d2b3e7a1d0ca1d221fe730f216b8715873/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp#L324-L341

The keys for the `topics` table would be `id`, `name`, `type`, `serialization_format`, and `offered_qos_profiles`. The keys for the `messages` table would be `id`, `topic_id`, `timestamp`, and `data`. Though, since this is split into multiple tables, we'd need to figure out how to make it generic for storage formats that might not use tables, so that the neutral intermediary format is truly neutral...

emersonknapp, 2021-04-07:
Ah - I see. I would say that - those keys those are implementation details of the Storage Implementation and need to stay fully hidden within the storage plugin, never to be exposed. What you're looking at there is, I think, misleading from the real data definition:

The `topics` Table is a "backup copy" of [`rosbag2_storage::BagMetadata`](https://github.com/ros2/rosbag2/blob/master/rosbag2_storage/include/rosbag2_storage/bag_metadata.hpp), which is fully stored in `metadata.yaml` as a storage-implementation-independend format (your "neutral intermediary format")- unless it has been lost and needs your Reindexer applied :). The `BagMetadata`, and there for the `Bag Metadata Version` that we mentioned before, contain all of this information. 

As for the `messages` table - this is a [`rosbag2_storage::SerializedBagMessage`](https://github.com/ros2/rosbag2/blob/master/rosbag2_storage/include/rosbag2_storage/serialized_bag_message.hpp). Now, the question is whether we need to version this structure. I think the answer is yes, because there are at least two features I can think of that we're considering that would need new fields. (source+received timestamp per message, and Publisher ID per message) - but again the fact that there are "database table keys" is an implementation detail of the storage implementation to be hidden, the important question is whether the storage implementation can read a given `SerializedBagMessageVersion`

jhdcs, 2021-04-07:
>  the important question is whether the storage implementation can read a given `SerializedBagMessageVersion`

This is why I was thinking of having the keys being in the neutral intermediary format (I'll just use NIF, it's shorter). Though the keys aren't going to be 1:1 name-wise. Mostly, I'd see it as a way to tell at-a-glance what type of information needs to be stored/read.

The check would work like: I get a Reader for format 1 (`Reader_1`) and a writer for format 2 (`Writer_2`). The reader reports that it would return NIF keys `X`, `Y`, and `Z`. The writer reports that it consumes `X`, `Y`, and `A`. So the conversion would occur with two warnings: One, that any data pertaining to `Z` would be lost. Two, that it had to use defaults for `A` in the new format.

The end user would not know what the actual internal storage names are for each of the formats. The NIF would describe them like "Message Timestamp", or "Message Index". If they went digging through the code, they could figure it out. But they could do that right now anyways.

At least, that's what I was thinking for that sort of a warning system.

emersonknapp, 2021-04-07:
I feel like I disagree, but I think I mainly am disagreeing with the way the concept is being presented. I would put it more like:

I have a `ReaderA`. `ReaderA` reports that has `SerializedBagMessage V1`. The current version of `SerializedBagMessage` is `V3` - according to the rosbag2 binaries that I am running. Because of version-management code, I know how to construct a `SerializedBagMessageV3` from a `SerializedBagMessageV1` - there is no "key juggling". Now, I can send these messages to the `WriterB`, which doesn't need to do anything special because it is receiving messages in the up-to-date format.

jhdcs, 2021-04-07:
The issue I see with that approach, though, is that you would need each version to understand how to convert their data to/from every other version. Any time someone adds a new version, you would have to go through all the previous versions and create code to convert to/from that version. If we convert to/from the NIF, we won't need to write any extra code. Everything else already knows how to extract what it needs from the NIF, and can report extra/missing values.

I could write out some Python pseudocode in a while, if you think that would help. Or I may be completely off-track here, and just making everything too complicated.

emersonknapp, 2021-04-07:
> Any time someone adds a new version, you would have to go through all the previous versions and create code to convert to/from that version

I'm not sure I see it this way. I think you only ever have to "convert up" - I don't think we need to promise that rosbags recorded on a newer version should be playable on an older version of rosbag2 (forward-compatibility, e.g play a Galactic-recorded bag using Foxy's rosbag2), only that rosbag2 is backwards-compatible. In this case, you basically only need conversion code like:

```
SerializedBagMessageV1();
SerializedBagMessageV2(SerializedBagMessageV1);
SerializedBagMessageV3(SerializedBagMessageV2);
// etc...
```

But instead of writing that, even - I think I'd prefer to find a more generic way that we could version schemas and translate between them. This would be useful for both `BagMetadata` and `SerializedBagMessage` schemas (which are the schemas of the 2 tables you're trying to think about here). There are tools for this, it's a common problem. I don't know if "schemaless key munging" is the approach I would prefer to take... The data feels more rigidly structured than that.

danthony06, 2021-04-07:
If I'm understanding the previous comment, are you saying we should not have the expectation to play a bag file recorded in a newer version of ROS when using an older version?

emersonknapp, 2021-04-07:
> If I'm understanding the previous comment, are you saying we should not have the expectation to play a bag file recorded in a newer version of ROS when using an older version?

That's what I was suggesting. It's what I have normally experienced in software data file versioning. Nothing's final, of course, if there's a strong argument and design to the contrary. If we used a library like Protobuf (just an example) for these schemas - then maybe we could offer both forward-and-backward compatibility.

jhdcs, 2021-04-07:
I had to familiarize myself with Protobuf, since I've not used it before. But it looks almost _exactly_ like something I was envisioning for the NIF...

danthony06, 2021-04-07:
It's a pretty frequent use case here that we run a mix of ROS and Ubuntu distributions, and frequently exchange data between the different systems. In some cases, we have vehicle hardware running specific versions of ROS that are extremely difficult or impossible to change. Not being able to open bag files generated from newer versions of ROS on an older version is going to cause us a lot of headaches.

emersonknapp, 2021-04-07:
> Not being able to open bag files generated from newer versions of ROS on an older version is going to cause us a lot of headaches.

It would be more helpful to provide an enumeration of actual use cases that can help inform a design, rather than dictating feature requirements or implementations. I am not saying that the requirement is wrong, just that the flow of information is backwards.

For example: do you anticipate that you want to generate J-turtle bags on your development/testing station, then copy those bag files over to a device that is pinned to H-turtle and play them back on the device? Or is the anticipated use case really that you want to play messages on a networked computer running J-turtle, so that the H-turtle device receives the messages? If it's the latter case, that's a different feature entirely that would need to be implemented in RMW.

danthony06, 2021-04-07:
There's two really common use cases I can think of.

1. We have internally run a variety of Ubuntu/ROS distributions, including 16.04, 18.04, and 20.04, which are all currently supported. We also interact with external engineers who may be running their own mix of Ubuntu versions. If we go out into the field, and use a laptop to bag some data from a live system that is experiencing issues, I expect to be able to be able to look at, and replay, the data in the field with my collaborators, even if they are running a different ROS version, as long as that ROS version is currently supported. I also expect to be able to bring that data to our lab, and distribute it to my colleagues, who can look at, and replay, that information, no matter what supported version of Ubuntu and ROS they are running on their laptop.

2. On single board computers or embedded systems like Nvidia's Jetson platforms, we might want to do hardware in the loop testing, using data recorded from a different computer, that is running a newer version of ROS. We may not have the network connectivity to stream data to these platforms, in the cases where they're operating in a big region, or are a small UAV.

All of these cases work in ROS 1. I don't remember ever having incompatible bag files, and I haven't experienced fundamental issues making different versions of ROS talk to each other, and my first extensive use of ROS 1 started with Fuerte.

danthony06, 2021-04-09:
Thinking about this some more, I think I incidentally hit on my real expectation, which is I want to use bags between any currently supported version of ROS. It's probably unreasonable to expect in the future that the B release of ROS be able to read things I generated in the T release, but if R, S, and T ROS releases are all currently supported, then I should be able to utilize all of them in a system, and exchange bags between them. There might even be a reasonable way of then warning users that when a distribution goes EOL, that it will not be able to read bags generated in future releases.

amacneil, 2021-04-09:
Serous question - Why is it "unreasonable to expect in the future that the B release of ROS be able to read things I generated in the T release"? It seems like a well defined bag format would not be changing so much. Many/most storage formats are able to go for a decade+ without breaking backwards compatibility.

danthony06, 2021-04-09:
I too would prefer that the bag format remain unchanged, and if that's not possible, I at least want to easily work with bags in currently supported versions of ROS.

Historically, in ROS 1, there were a few version changes, and no explicit guarantee of compatibility: http://wiki.ros.org/Bags/Format. It's been stable for quite some time. I don't want to completely derail @jhdcs's issue, so maybe there is a more appropriate place to continue this discussion.

jhdcs, 2021-04-12:
Reading Emerson's comment at [here](https://github.com/ros2/rosbag2/pull/709#discussion_r611110847), perhaps  we should make some clearer definitions for each of the operations that the converter can perform?

Some ideas:
Changing storage version -> `re-version`
Changing storage implementation (ex: from `sqlite` to `ros1bag`) -> `transmute`

emersonknapp, 2021-04-12:
> Reading Emerson's comment at [here](https://github.com/ros2/rosbag2/pull/709#discussion_r611110847), perhaps we should make some clearer definitions for each of the operations that the converter can [...]

I don't mind if they're a little more verbose and therefore self-contained definitions - especially since it sounds like these should mostly be "under the hood". I'd rather wait until later - I am seeing something like the following concepts:

* "Bag Forward/Back Compatibility" - storage reads past and future versions as well as they can, by having a schema definition with forward/backward compatibility in mind. I see that this breaks down to the following two, ideally which have a similar solution
  * "Bag Metadata Schema Compatibility" for metadata.yaml and/or the metadata as stored in the bagfile
  * "Message Schema Compatibility" for the `SerializedBagMessage` which is the schema for how messages are stored (currently `timestamp,topic,serialized_data`)
* "Bag Rewriting" - opening a bag with one `StorageOptions+RecordOptions` and write it out with a different set (feature superset over "changing storage implementation") - happening at #709
* "Message Definition Flexibility" - by having the message definition (`.msg`/`IDL`) stored in the bag, be able to change a message, before publishing, to some receiver's current version of that definition. This one I think needs a lot more independent hashing out - the way it worked in ROS1 was very different than how the RMW/DDS API works now.

jhdcs, 2021-07-30:
This is being closed in favor of other spawned issues.

closed 2021-07-30 by jhdcs