ros2/rosbag2 #720: Remove `-Werror` from CMakeLists.txt

Karsten1987, 2021-04-08:
While it is generally appreciated to compile without warnings, there are situations where we don't want to fail the compilation on the first warning, such as a buildfarm run or trying to (cross-)compile rosbag2 on different platforms and OS's.

One can still compile locally with `-Werror` enabled by passing in the respective cmake args to `colcon` and their-like.

emersonknapp, 2021-04-08:
Hmm - I like `-Werror` "for real", but it is sometimes annoying while developing. I'd be inclined to enable it via Action CI to gate PRs.

I agree that it is better for the nightly to go yellow rather than red when there's a warning, though.

clalancette, 2021-04-08:
@Karsten1987 Do you mind making a PR for this?  Combining the original suggestion with @emersonknapp's suggestion, we should probably remove `-Werror` from all of the CMakeLists.txt in this repository, and then add a `--cmake-args -DCMAKE_CXX_FLAGS="-Werror"` to https://github.com/ros2/rosbag2/blob/master/.github/workflows/test.yml (I'm pretty sure I added support for arbitrary `colcon` command-line arguments to `action-ros-ci`, though I can't quite remember now).

emersonknapp, 2021-04-08:
On it #722

PR #722 (2021-04-08): Remove -Werror from CMakeLists.txt, enable it in Action CI
Closes #720
closed 2021-04-08 by emersonknapp