ros2/rosbag2 #732: test_record, test_record_regex, test_record_all are flaky

ivanpauno, 2021-04-13:
## Description
** clear and concise description of the bug is **

The three mentioned tests sometimes fail in CI jobs:

* https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2312/#showFailuresLink

### Expected Behavior
** clear and concise description of what you _expected_ to happen **

Shouldn't fail.

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

The sometimes fail, particularly in the "repeated" jobs.

## System (please complete the following information)
 - OS: macOS, might affect other OSs as well
 - ROS 2 Distro: Rolling
 - Version: https://github.com/ros2/rosbag2/commit/453281c058505c1d39218e5498096dc9e57d8a5e

## Additional context
** Add any other context about the problem here **

emersonknapp, 2021-04-13:
Yeah - these tests have been a real problem. I'm hoping that some of the refactoring work we are doing right now will make it a lot easier to construct these tests in a reliable way. We've got a few similar (but not identical) tickets in the backlog - I'd like to try and make a "code/test quality burndown board" to focus on when we feature-freeze rosbag2 on May 1.

tfoote, 2021-05-13:
Also linux with cyclonedds_cpp is now failing more often than passing on repeated: https://ci.ros2.org/job/nightly_linux-aarch64_repeated/1593/testReport/(root)/projectroot/test_record__rmw_cyclonedds_cpp/history/

[image]

Also related #766

tfoote, 2021-07-30:
More instances here: 

* http://build.ros2.org/job/Fci__nightly-release_ubuntu_focal_amd64/295/testReport/
* 

Also potentially test_play
* http://build.ros2.org/job/Rci__nightly-debug_ubuntu_focal_amd64/395/testReport/
* https://ci.ros2.org/job/nightly_linux_repeated/2363/testReport/junit/(root)/projectroot/test_play__rmw_fastrtps_cpp/

sloretz, 2021-09-20:
> Also potentially test_play

[Another example](https://ci.ros2.org/job/nightly_linux_repeated/2416) of `test_play` being flaky on Ubuntu Focal with FastDDS

The test timed out while running `RosBag2PlayTestFixture.recorded_messages_are_played_for_filtered_topics`

```
projectroot.test_play__rmw_fastrtps_cpp
rosbag2_transport.test_play__rmw_fastrtps_cpp.gtest.missing_result
```

<details><summary>Console output</summary>

```
      Start 29: test_play__rmw_fastrtps_cpp

29: Test command: /home/jenkins-agent/workspace/nightly_linux_repeated/venv/bin/python3.8 "-u" "/home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_play__rmw_fastrtps_cpp.gtest.xml" "--package-name" "rosbag2_transport" "--output-file" "/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/ament_cmake_gmock/test_play__rmw_fastrtps_cpp.txt" "--env" "RMW_IMPLEMENTATION=rmw_fastrtps_cpp" "--command" "/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/test_play__rmw_fastrtps_cpp" "--gtest_output=xml:/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_play__rmw_fastrtps_cpp.gtest.xml"
29: Test timeout computed to be: 60
29: -- run_test.py: extra environment variables:
29:  - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
29: -- run_test.py: invoking following command in '/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport':
29:  - /home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/test_play__rmw_fastrtps_cpp --gtest_output=xml:/home/jenkins-agent/workspace/nightly_linux_repeated/ws/build/rosbag2_transport/test_results/rosbag2_transport/test_play__rmw_fastrtps_cpp.gtest.xml
29: Running main() from gmock_main.cc
29: [==========] Running 9 tests from 2 test suites.
29: [----------] Global test environment set-up.
29: [----------] 4 tests from RosBag2PlayTestFixture
29: [ RUN      ] RosBag2PlayTestFixture.recorded_messages_are_played_for_all_topics
29: [       OK ] RosBag2PlayTestFixture.recorded_messages_are_played_for_all_topics (982 ms)
29: [ RUN      ] RosBag2PlayTestFixture.recorded_messages_are_played_for_all_topics_with_unknown_type
29: [WARN] [1632150895.972629901] [rosbag2_player]: Ignoring a topic 'topic3', reason: package 'unknown_msgs' not found, searching: [/home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_transport, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_compression, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_storage_default_plugins, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_storage, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_test_common, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rclcpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/libstatistics_collector, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rcl, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_implementation, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_fastrtps_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/test_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_connextddsmicro, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_connextdds, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/std_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/statistics_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosgraph_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosbag2_interfaces, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_fastrtps_shared_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_cyclonedds_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_connextdds_common, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rcl_interfaces, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/action_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/unique_identifier_msgs, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rmw_dds_common, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/builtin_interfaces, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_default_runtime, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_typesupport_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_generator_py, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_typesupport_c, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_typesupport_fastrtps_c, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_typesupport_introspection_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rosidl_typesupport_fastrtps_cpp, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rcl_yaml_param_parser, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/rcl_logging_spdlog, /home/jenkins-agent/workspace/nightly_li [... cut ...] linux_repeated/ws/install/ament_package, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/ament_lint, /home/jenkins-agent/workspace/nightly_linux_repeated/ws/install/ament_cppcheck].
29: [       OK ] RosBag2PlayTestFixture.recorded_messages_are_played_for_all_topics_with_unknown_type (972 ms)
29: [ RUN      ] RosBag2PlayTestFixture.recorded_messages_are_played_for_filtered_topics
    Test #29: test_play__rmw_fastrtps_cpp ........................***Timeout  60.01 sec
```
</details>

clalancette, 2022-04-13:
https://github.com/ros2/rosbag2/pull/988 should improve the situation, I believe.

MichaelOrlov, 2024-05-05:
- Closing as stale issue.

closed 2024-05-05 by MichaelOrlov