ros2/rosbag2 #733: Build failed on  Ubuntu bionic

SandyWare, 2021-04-16:
When I try to build and install ROS Bag2 using  
   $ colcon build --merge-install

It gives me the below error 
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_test_common
Starting >>> zstd_vendor
--- stderr: shared_queues_vendor
CMake Error at CMakeLists.txt:56 (ament_export_targets):
  Unknown CMake command "ament_export_targets".
---
Failed   <<< shared_queues_vendor [25.7s, exited with code 1]
Aborted  <<< rosbag2_test_common [28.7s]
Aborted  <<< zstd_vendor [30.7s]
Aborted  <<< sqlite3_vendor [32.6s]

Summary: 0 packages finished [33.1s]
  1 package failed: shared_queues_vendor
  3 packages aborted: rosbag2_test_common sqlite3_vendor zstd_vendor
  2 packages had stderr output: rosbag2_test_common shared_queues_vendor
  12 packages not processed

## System (please complete the following information)
    OS: Ubuntu 18.04.5 LTS (bionic)
    ROS 2 Distro: ROS Crystal.
    Python 3.6.9
 
## Additional context
 I am trying to load ROSBag in python read rosbag data.
   import rosbag

Any idea what is missing from my end?

vinnnyr, 2021-04-16:
@SandyWare Can you confirm if you have tried to checkout the `crystal` branch since you mentioned that is the distro you are using? (https://github.com/ros2/rosbag2/tree/crystal)

I believe the `master` branch is only meant to build from `rolling`

emersonknapp, 2021-04-16:
> git clone https://github.com/ros2/rosbag2.git

This checks out the default branch `master`, which indeed only is intended to build against the latest source from github.com/ros2/ros2 - there not even a strong guarantee that it will build against installed Rolling distro

closed 2021-04-16 by emersonknapp

SandyWare, 2021-04-17:
Thanks for the suggestion.  I try to build with crystal branch.
$ git clone -b crystal https://github.com/ros2/rosbag2.git
$ colcon build --merge-install

Nope, I am still getting below erorr : 

Starting >>> rosbag2_test_common
Starting >>> shared_queues_vendor
Starting >>> ros1_rosbag_storage_vendor
Starting >>> sqlite3_vendor
[Processing: ros1_rosbag_storage_vendor, rosbag2_test_common, shared_queues_vendor, sqlite3_vendor]
Finished <<< rosbag2_test_common [40.4s]
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [40.9s]
Finished <<< sqlite3_vendor [52.7s]
Finished <<< ros1_rosbag_storage_vendor [55.8s]
[Processing: rosbag2_storage]
[Processing: rosbag2_storage]
Finished <<< rosbag2_storage [1min 24s]
Starting >>> rosbag2
Starting >>> rosbag2_storage_default_plugins
--- stderr: rosbag2
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< rosbag2 [34.9s, exited with code 1]
[Processing: rosbag2, rosbag2_storage_default_plugins]
Aborted  <<< rosbag2_storage_default_plugins [40.6s]

Summary: 5 packages finished [2min 46s]
  1 package failed: rosbag2
  1 package aborted: rosbag2_storage_default_plugins
  1 package had stderr output: rosbag2
  5 packages not processed

vinnnyr, 2021-04-17:
@SandyWare this is a different build error. try using rosdep to install dependencies.

You can also try installing this particular dependency directly as an deb : sudo apt-get install ros-crystal-test-msgs