ros2/rosbag2 #734: Segmentation fault when trying to record with `--qos-profile-overrides-path`

clalancette, 2021-04-16:
## Description
In terminal 1, I'm running:

```
ros2 topic pub -r 0.1 --qos-durability transient_local /talker std_msgs/String "data: Hello World"
```

(that is, publish to /talker every 10 seconds with `transient_local` durability)

I also have the follow QoS override file (`durability_override.yaml`):

```
/talker:
  durability: transient_local
  history: keep_all
```

In terminal 2 I run:

```
ros2 bag record -a -o my_bag --qos-profile-overrides-path durability_override.yaml
```

When I run that `ros2 bag record` command, I'm getting a segmentation fault.

### Expected Behavior

`ros2 bag record` runs and records data.

### Actual Behavior

Segmentation fault.

## System (please complete the following information)
 - OS: arm64 Ubuntu Focal (20.04)
 - ROS 2 Distro: Rolling
 - Version: ros-rolling-rosbag2* 0.7.0 Debian packages

## Additional context

I haven't yet had time to debug this any further than getting the crash.

emersonknapp, 2021-04-16:
Thanks for reporting this - I'll look at it ASAP

fujitatomoya, 2021-04-19:
@clalancette @emersonknapp 

i cannot reproduce a segmentation fault with https://github.com/ros2/ros2/commit/2c3dcb37040365ee5f09eede264cd05910134076.

```
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 bag record -a -o my_bag --qos-profile-overrides-path durability_override.yaml
[INFO] [1618810451.971882969] [rosbag2_storage]: Opened database 'my_bag/my_bag_0.db3' for READ_WRITE.
[INFO] [1618810451.971959660] [rosbag2_recorder]: Listening for topics...
[INFO] [1618810451.972392254] [rosbag2_recorder]: Overriding subscription profile for /talker
[INFO] [1618810451.973413563] [rosbag2_recorder]: Subscribed to topic '/talker'
[INFO] [1618810451.974541773] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1618810451.975995836] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[WARN] [1618810451.976288939] [rosbag2_recorder]: A new publisher for susbcribed topic /parameter_events was found offering RMW_QOS_POLICY_DURABILITY_VOLATILE, but rosbag2 already subscribed requesting RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL. Messages from this new publisher will not be recorded.
^C[INFO] [1618810562.865246814] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1618810563.169081029] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 bag info ./my_bag/

Files:             my_bag_0.db3
Bag size:          18.0 KiB
Storage id:        sqlite3
Duration:          103.882s
Start:             Apr 19 2021 14:34:11.976 (1618810451.976)
End:               Apr 19 2021 14:35:55.858 (1618810555.858)
Messages:          17
Topic information: Topic: /parameter_events | Type: rcl_interfaces/msg/ParameterEvent | Count: 0 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 6 | Serialization Format: cdr
                   Topic: /talker | Type: std_msgs/msg/String | Count: 11 | Serialization Format: cdr
```

clalancette, 2021-04-19:
@fujitatomoya Ah, very interesting.  I also cannot reproduce if I build from source.  However, if I use the debian packages, I can reproduce.  I then went back to tag 0.7.0 for rosbag2 when building from source, and I can reproduce it again.

So @emersonknapp it looks like your job is pretty easy; if you do a new tag-and-release of rosbag2, this issue will probably be fixed.  Alternatively, just let me know and I can do the release for you.  Thanks!

emersonknapp, 2021-04-19:
Sure - I'll release a new version right away. Thanks for digging in @fujitatomoya and @clalancette

emersonknapp, 2021-04-19:
Whoops, github comment language can be tricky - will not close until bloom is done

emersonknapp, 2021-04-19:
https://github.com/ros/rosdistro/pull/29206

clalancette, 2021-04-19:
I can confirm that with that rosdistro PR merged, this is no longer an issue.  So I'm going to close this out, thanks @emersonknapp !

PR #738 (2021-04-19): Version 0.8.0
Partly fixes #734
closed 2021-04-19 by clalancette