ros2/rosbag2 #756: ros2 bag play does not work with rmw_connextdds

hidmic, 2021-04-23:
## Description
Basic `ros2 bag play` usage fails when `rmw_connextdds` is chosen as `RMW_IMPLEMENTATION`. This seems to be the case for the simplest tutorial (i.e. [Recording and playing back data](https://docs.ros.org/en/rolling/Tutorials/Ros2bag/Recording-And-Playing-Back-Data.html)).

### Expected Behavior
Recording and playing back bagfiles while using `rmw_connextdds` as `RMW_IMPLEMENTATION` works fine.

### Actual Behavior

`ros2 bag record` _seems_ to work, but then `ros2 bag play` fails with:

```sh
> ros2 bag play subset
[INFO] [1619194035.972364200] [rosbag2_storage]: Opened database 'subset\subset_0.db3' for READ_ONLY.
[D0143|Pub(80000008)|T=rt/turtle1/cmd_vel|CREATE Writer]DDS_DeadlineQosPolicy_is_consistentI:inconsistent QoS policy: period
[D0143|Pub(80000008)|T=rt/turtle1/cmd_vel|CREATE Writer]DDS_DataWriterQos_is_consistentI:inconsistent QoS policy: deadline
[D0143|Pub(80000008)|T=rt/turtle1/cmd_vel|CREATE Writer]DDS_Publisher_create_datawriter_disabledI:ERROR: Inconsistent QoS
[ERROR] [1619194035.979007400] [rmw_connextdds]: failed to create DDS writer
[ERROR] [1619194035.982137900] [rmw_connextdds]: failed to allocate RMW_Connext_Publisher
[ERROR] [1619194035.984866600] [rmw_connextdds]: failed to create RMW publisher
[WARN] [1619194035.987242500] [rosbag2_player]: Ignoring a topic '/turtle1/cmd_vel', reason: could not create publisher: error not set, at C:\opt\ros\rolling_src\src\ros2\rcl\rcl\src\rcl\publisher.c:115.
[D0143|Pub(80000008)|T=rt/turtle1/pose|CREATE Writer]DDS_DeadlineQosPolicy_is_consistentI:inconsistent QoS policy: period
[D0143|Pub(80000008)|T=rt/turtle1/pose|CREATE Writer]DDS_DataWriterQos_is_consistentI:inconsistent QoS policy: deadline
[D0143|Pub(80000008)|T=rt/turtle1/pose|CREATE Writer]DDS_Publisher_create_datawriter_disabledI:ERROR: Inconsistent QoS
[ERROR] [1619194035.997205400] [rmw_connextdds]: failed to create DDS writer
[ERROR] [1619194035.999492900] [rmw_connextdds]: failed to allocate RMW_Connext_Publisher
[ERROR] [1619194036.002531900] [rmw_connextdds]: failed to create RMW publisher
[WARN] [1619194036.004780000] [rosbag2_player]: Ignoring a topic '/turtle1/pose', reason: could not create publisher: error not set, at C:\opt\ros\rolling_src\src\ros2\rcl\rcl\src\rcl\publisher.c:115.
```

## To Reproduce

1. Simply follow the [Recording and playing back data](https://docs.ros.org/en/rolling/Tutorials/Ros2bag/Recording-And-Playing-Back-Data.html) tutorial.

## System 
 - OS: Windows 10, Ubuntu Focal 20.04
 - ROS 2 Distro: Rolling
 - Version: cdbee9bc3dfe23bada4b5a7276b595dc17a8fc3a

## Additional considerations

This may as well be a problem in [ros2/rmw_connextdds](https://github.com/ros2/rmw_connextdds/).

emersonknapp, 2021-04-23:
Taking a look today - this looks related to the `RMW_DURATION_INFINITE` logic - specifically on the Deadline policy.

emersonknapp, 2021-04-23:
@asorbini - can I ask you to migrate that change into the new implementation? I don't think I have time to learn the structure of the new package at the moment, there are other Galactic release issues I need to take care of

asorbini, 2021-04-23:
@emersonknapp I will take a look and port the changes.

asorbini, 2021-04-23:
ros2/rmw_connextdds#43 should resolve the issue

emersonknapp, 2021-05-14:
This was resolved by ros2/rmw_connextdds#43

closed 2021-05-14 by emersonknapp