ros2/rosbag2 #758: Cut new release for rolling & galactic

Karsten1987, 2021-04-26:
In order to make the [buildfarm pass rosbag2_bag_v2](https://build.ros2.org/view/Gbin_uF64/job/Gbin_uF64__rosbag2_bag_v2_plugins__ubuntu_focal_amd64__binary/), we needs a new release for rolling & galactic. 

@emersonknapp I'd propose to cut a new release after we merged https://github.com/ros2/rosbag2/pull/757. Is there anything related to clock which should land before? Obviously, that's not a final galactic release, but we can get things started.

emersonknapp, 2021-04-26:
I would also like to land https://github.com/ros2/rosbag2/pull/754 before cutting a release

emersonknapp, 2021-04-26:
@Karsten1987 could use your review on that #754

emersonknapp, 2021-05-17:
Closing this with
* rolling: https://github.com/ros/rosdistro/pull/29556
* galactic: https://github.com/ros/rosdistro/pull/29555

closed 2021-05-17 by emersonknapp