ros2/rosbag2 #764: Fill SerializedBagMessage.time_stamp with simulation time when recording a bag from a simulation

doisyg, 2021-05-02:
## Description

It seems that when recording (with` ros2 bag record` cli) a bag of a ros2 app running from simulation (and with a /clock topic), the SerializedBagMessage.time_stamp is set to the system time and not the simulation /clock time.
It would be convenient to fix in order not to run into issues of time reference when replaying a bag recorded from a simulation.
(or maybe it is already the case through an arg/config that I missed)

## Implementation Notes / Suggestions
I don't know what's the code path from the record cli, but if the underlying recording process is rclcpp::Node, I guess it is just a matter of switching its use_sim_time parameter to True

emersonknapp, 2021-05-04:
Thanks for this request. This won't be done for the Galactic release, but we can try to think about if this could be backportable to Foxy and Galactic without breaking API/ABI.

salmagro, 2021-05-04:
This option will be really convenient. I appreciate any improvement on that.

emersonknapp, 2021-05-21:
Related to https://github.com/ros2/rosbag2/issues/696 - once we know what the simulation time is, it should be easy to store it on messages

Adum888, 2022-03-30:
Hi are there any updates on this topic?
I'm runnig into trouble when merging to bagfiles that have data derived from the same source bag file (played with --clock). The Timestamps (in the header) seem to be alright (-->sim-time) but the placement in the new bagfile is relative to the recording time

MichaelOrlov, 2022-04-01:
@Adum888 @emersonknapp As far as I am understand correctly this feature has not been done yet.
It looks like implementation is going to be pretty straight forward.
Will need to change one line in [Recorder::create_subscription(..)](https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L261) callback to start using clock from node. e.g.
```cpp
writer_->write(message, topic_name, topic_type, this->get_clock()->now());
```
and somewhere in Recorder constructor add initialization for node to use sim time.
```cpp
  this->set_parameter(rclcpp::Parameter("use_sim_time", record_options_.use_sim_time));
```
Of course will need to add appropriate `record_options_.use_sim_time` and command line argument for `record` verb.

KeisukeShima, 2022-04-26:
@MichaelOrlov @emersonknapp 
I have created a PR based on your comments, would you review the PR?
https://github.com/ros2/rosbag2/pull/994

PR #994 (2022-05-12): Add use_sim_time option to record verb
Closes #764 

Based on @MichaelOrlov  's ([comment](https://github.com/ros2/rosbag2/issues/764#issuecomment-1085312999)), I have created a PR to reflect use_sim_time in the record verb.

## Test procedure

Before testing, please prepare a bag file.

* Terminal 1
```sh
ros2 bag play --clock 10 ORIGINAL_BAG
```

* Terminal 2
```sh
ros2 bag record --use-sim-time -a -o RECORDED_BAG
```

Then compare the values of `/clock`. They should be almost the same.
ex.

* Terminal 1
```sh
ros2 bag play --clock 10 [ORIGINAL_BAG|RECORDED_BAG]
```

* Terminal 2
```sh
ros2  [... cut]
closed 2022-05-12 by MichaelOrlov