ros2/rosbag2 #766: records_sensor_data test is flaky

tfoote, 2021-05-10:
## Description

records_sensor_data test fails about once per week in nightly builds: https://ci.ros2.org/job/nightly_linux-aarch64_repeated/1590/testReport/rosbag2_transport/RecordIntegrationTestFixture/records_sensor_data/history/

[image]

### Expected Behavior
It should pass every nightly.

### Actual Behavior
Be repeatable. 

## To Reproduce
Run the CI repeated nightly tests: https://ci.ros2.org/job/nightly_linux-aarch64_repeated/

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: Nightly
 - Version: head

## Additional context
* From the buildfarmer monitoring.

Barry-Xu-2018, 2021-06-29:
This problem maybe similar to #790.   
While publisher sends messages, subscription of rosbag may not connect to publisher.    

Unfortunately, I cannot reproduce this after run `test_record` 100 times in my local environment.  
After fixing #790, I can update test codes of `test_record`.

MichaelOrlov, 2023-04-13:
- Closing as not reproducible any more
[image]

closed 2023-04-13 by MichaelOrlov