ros2/rosbag2 #772: Incorrect QoS compatibility warning

eboasson, 2021-05-20:
## Description

Sometimes `ros2 bag record` erroneously reports a volatile publisher is discovered but rosbag is recording with transient-local QoS already.

### Expected Behavior

No warning. The subscription taken by rosbag is volatile as it should be.

### Actual Behavior

It reports this:
```
A new publisher for susbcribed topic /turtle1/cmd_vel was found offering RMW_QOS_POLICY_DURABILITY_VOLATILE, but rosbag2 already subscribed requesting RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL. Messages from this new publisher will not be recorded.
```
the subscription taken is actually volatile.

## To Reproduce

1. `ros2 ros2 run turtlesim turtle_teleop_key`
2. `ros2 bag record /turtle1/cmd_vel`

It doesn't always go wrong, it seems to be timing-dependent and require a release build.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: what is about to be released as Galactic
 - Version: source build using https://github.com/ros2/ros2/commit/052d7c3b8ac3da36d7e89725564abd6c78ada682

## Additional context

A bit of digging suggests the problem is in `rosbag2_transport/src/rosbag2_transport/recorder.cpp`:
```C++
void Recorder::warn_if_new_qos_for_subscribed_topic(const std::string & topic_name)
{
  auto existing_subscription = subscriptions_.find(topic_name);
  if (existing_subscription == subscriptions_.end()) {
    // Not subscribed yet
    return;
  }
  if (topics_warned_about_incompatibility_.count(topic_name) > 0) {
    // Already warned about this topic
    return;
  }
  const auto & used_profile = existing_subscription->second->get_actual_qos().get_rmw_qos_profile();
```
I think (but I am no C++ expert by any stretch of the imagination) that the temporary `rclcpp::QoS` object returned by `get_actual_qos()` is not kept alive long enough and that the memory pointed to by `used_profile` gets reused.

eboasson, 2021-05-21:
I agree race conditions involving `subscriptions_` likely exist, but in this case I don't think that is actually a problem:
* there is only one specific topic being recorded, so `Recorder::record` subscribes to it in the call to `subscribe_topics`;
* somewhere in that call the newly created subscription is added to `subscriptions_`;
* at this point all subscriptions ever to be taken are present, and I would therefore expect `subscriptions_` to never be modified during the recording;
* the discovery thread is started immediately following that call to `subscribe_topics`, so it should observe the subscription being in `subscriptions_`;
* therefore when it discovers the publisher, it won't create a new subscription but only check the QoS.

This is consistent with the observed behaviour: always one subscription, with the correct QoS and a warning depending on "circumstances".

Naturally, this line of reasoning falls apart as soon as the discovery thread has actual work to perform.

If in this particular scenario the above line of reasoning is indeed valid, then it can't have been a race condition. Given that `get_actual_qos` returns a temporary `rclcpp::QoS` and `get_rmw_qos_profile` then returns a reference into this temporary (but without this being visible to the compiler at this point, so all it has is a random pointer to somewhere), I don't see why the lifetime of that `rclcpp::QoS` object would extend beyond the statement.

I suspect the timing dependency is because the discovery thread may (or may not) stop before it discovers the presence of a publisher, which ultimately involves DDS discovery and so is timing sensitive.

Incidentally, I think it somewhat odd that the discovery thread monitors for such QoS issues, but stops doing that once it has created all required subscriptions. So if I create a publisher with an incompatible QoS early on, I get a warning; but if I do it later, it don't get one. That's inconsistent and makes the warning rather useless.

emersonknapp, 2021-05-21:
I see what you mean about the object lifetime and the reference - do you think that changing to the following would fix it? Is there a way we could robustly test this? (I would be fine with merging it without a regression test, as it's obviously an improvement, but it would be nice to find a way to prove it, if possible)

```
const auto actual_qos = existing_subscription->second->get_actual_qos();
const auto & used_profile = actual_qos.get_rmw_qos_profile(); 
```

As for the other problems - it would be good to address:
* synchronized access to `subscriptions_`
* run discovery thread forever in order to notice new incompatible publishers on a topic?

emersonknapp, 2021-06-30:
@eboasson can you try out #803 to see if it solves the problem for you? I'm not sure how to regression test this, since it's undefined behavior and isn't always reproducible - any ideas on that?

eboasson, 2021-07-02:
@emersonknapp it solves it for me, with the original code it was 100% reproducible and with #803 added I've been unable to reproduce it.

I wouldn't try to regression test for it: that'd mean something like requiring a build on a Linux/aarch64 with gcc version `9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)` and then doing a before-and-after test. After all, any gcc patch release could change the behaviour ... To me it makes more sense to simply say "case closed".

PR #803 (2021-07-02): Copy recorder QoS profile to local variable so that temporary value isn't cleared
Fixes #772
Fixes #802

This function was using a temporary variable that got cleared, and so had undefined behavior. Create a local variable to make the reference valid.
closed 2021-07-02 by emersonknapp