ros2/rosbag2 #777: Finalize officially-supported `foxy-future` branch

emersonknapp, 2021-06-04:
## Description

As part of the #657 discussion, we've realized we cannot do a full backport in a stable way.

Stabilize the exact commit and modifications for the `foxy-future` branch.

## Related Issues

#657 

## Completion Criteria

* A `foxy-future` branch exists
* It contains a version of rosbag2 closely corresponding to Spring 2021 latest development, including many performance improvements and some new features
* It builds against the released Foxy binaries
* It is built on a regular schedule automatically to ensure stability
* The `ros2 bag` command for this version prints a notice "You are using Rosbag2 Foxy-future. This is an advanced alternate distribution for Foxy with performance improvements." - this notice gives users confidence that they are using the right version
* The README contains details about the "foxy-future" meaning, support, and how to use it (build it from source in your workspace)

## Implementation Notes / Suggestions

See completion criteria

## Testing Notes / Suggestions

See completion criteria

JWhitleyWork, 2021-06-21:
Can this branch now be `bloom-release`'d as `rosbag2-future`?

emersonknapp, 2021-06-21:
No - there will not be a bloom release because there isn't a way to resolve the conflicts between the binary Foxy release and the branch https://github.com/ros2/rosbag2/tree/foxy-future. You use the branch by building it from source in a workspace, against the Foxy core.

Closing this out as the branch is now finalized.

PR #687 (2021-06-18): "Futurize" foxy-future branch
Resolves #777 
Part of #657 

Notes:
* Some copy-pasted code from the Galactic core that is not present in Foxy for the utilities in here.
* Some minor API tweaks for things that changed since Foxy
* Add information to the README about this special branch
* Add a notice to the ros2bag CLI that informs you that you're using this version

See https://discourse.ros.org/t/fast-forward-merging-rosbag2-master-api-to-foxy/18927/38 for more context
closed 2021-06-21 by emersonknapp