ros2/rosbag2 #781: rosbag2 Cookbook (Programmatic Interface Examples)

danthony06, 2021-06-15:
## Description

The original rosbag had a set of documentation, called the rosbag [Cookbook](http://wiki.ros.org/rosbag/Cookbook). It contains a lot of examples of how to programmatically interact with bags to accomplish common tasks, like post-processing bag data and fixing some common issues with bags. While looking at https://github.com/ros2/rosbag2/pull/755, I was searching for similar documentation on how to interact with a ROS 2 bag, and I couldn't find it. Maybe I'm just looking in the wrong place, but it seems valuable to have. The existing documentation seems to cover the command line interface for recording and playing back bags, but not really just manipulating them through an API.

## Related Issues

None

## Completion Criteria

- An example of opening and reading a specific set of topics back from an existing bag
- An example of opening a new bag, creating some kind of message, and writing it back to the bag
- An example of reading the metadata out of a bag file

## Implementation Notes / Suggestions

We could put this in a separate repo, but I think it makes sense to have the documentation live somewhere in this repo.

## Testing Notes / Suggestions

This is intended to be documentation, not adding additional code.

emersonknapp, 2021-06-15:
Thanks for the ticket! This is something that @gbiggs is currently looking at - we discussed it in the last Tooling WG meeting.

Considering that we don't have a ROS 2 Wiki (yet), we're still not entirely sure where these should go, but we're leaning towards just adding a sub-page to http://docs.ros.org/en/foxy/Tutorials.html as a start. Maintaining a site like navigation.ros.org would be another possible alternative, but we'd rather not own that infrastructure if we don't have to.

gbiggs, 2021-06-15:
You can take a look at https://github.com/ros2/ros2_documentation/pull/1663 to see the first tutorial's draft.

danthony06, 2021-06-16:
@gbiggs that's close, but I'm thinking more along the lines of tools where I don't even have to launch a ROS 2 system. For instance, quite frequently I write short little Python scripts that open up a bag file, runs through the data, and performs some little simple analysis. Or it could even be something as simple as opening up a bag file, and converting raw stereo images to point clouds or disparity images to avoid the overhead and time delays of trying to do that through a ROS 2 system.

danthony06, 2021-06-16:
An interface more like this:
https://gitlab.com/ternaris/rosbags

gbiggs, 2021-06-16:
That sort of tutorial is on the list to be written. I can't say when it will happen, though.

emersonknapp, 2023-07-07:
Now that the examples packages are written and released (https://github.com/ros2/rosbag2/tree/rolling/rosbag2_examples), I'm going to close this issue as complete. We can reopen new issues with more specific asks if necessary

closed 2023-07-07 by emersonknapp