ros2/rosbag2 #786: Delay option for ros2 bag play

hayato-m126, 2021-06-22:
I want to use the delay option of bag play that was implemented in ROS1, but it seems that it is not implemented in ROS2.
Is there any reason why it is not implemented in ROS2?

TierIV implemented the delay option in the forked repository below, if we submit a pull request to this official repository, will it be accepted?

https://github.com/tier4/rosbag2/pull/3/files

emersonknapp, 2021-06-22:
> Is there any reason why it is not implemented in ROS2?

No reason, just hasn't been built

> if we submit a pull request to this official repository, will it be accepted?

Assuming we iron out anything that comes up during code review, yes!

Note: i've edited your original ticket (leaving all the original context) to lay out the feature request. Please check that it matches what you have in mind.

hayato-m126, 2021-06-23:
Thank you for reply.
The edited issue is as I intended.

MichaelOrlov, 2021-07-02:
@hayato-m126 Could you please clarify what a use cases for usage this `delay` API?

hayato-m126, 2021-07-02:
@MichaelOrlov 
The delay option is assumed to be used when calling ros bag play from ros launch.
When ros bag play is launched with multiple nodes such as sensing, percepiton, localization, etc., sometimes data be output before each node is fully initialized, so this option is used to delay data playback.

MichaelOrlov, 2021-07-02:
@hayato-m126 I would ecomend to implement `wait_for_matched` API for Player class instead for your cases.

hayato-m126, 2021-07-02:
@MichaelOrlov 
Do you mean to implement an option to wait for playback based on generic conditions, not just time?
If so, what are the possible use cases for using conditions other than time?

Since delay is an option that existed in ROS1, Tier4 simply implemented the same functionality in the pull request associated with this issue.

MichaelOrlov, 2021-07-03:
@hayato-m126 I understand that you are trying to repeat functionality from ROS1.
However there are exists possibility to wait for matched number of subscribers for publisher rather than just blind delay.
i.e. Player will not start playback until publisher will will discover number of specified subscribers. It means until subscribers nodes will be initialized and visible for publisher via discovery.
We have implemented some similar functionality for unit tests. Please refer to the  [wait_for_matched in PublicationManager](https://github.com/ros2/rosbag2/blob/master/rosbag2_test_common/include/rosbag2_test_common/publication_manager.hpp#L98-L152)

However usage of the `wait_for_matched` API for Player class via ros launcher might require to pass many parameters like number of matching nodes per topic and timeout. Maybe need to consider to pass them via some config file. 
I would assume that `wait_for_matched` would fully replace `delay` feature.

hayato-m126, 2021-07-04:
@MichaelOrlov 
I read the unit test code you showed me.

I understand that wait_for_matched is suitable for the purpose of waiting for the subscriber to be ready, although the way of specifying the condition is a bit complicated.

Thank you for your advice.

hayato-m126, 2021-07-07:
I close this issue because the linked PR has been merged.

closed 2021-07-07 by hayato-m126

yinhaosen, 2023-07-28:
> @hayato-m126 I understand that you are trying to repeat functionality from ROS1. However there are exists possibility to wait for matched number of subscribers for publisher rather than just blind delay. i.e. Player will not start playback until publisher will discover number of specified subscribers. It means until subscribers nodes will be initialized and visible for publisher via discovery. We have implemented some similar functionality for unit tests. Please refer to the [wait_for_matched in PublicationManager](https://github.com/ros2/rosbag2/blob/master/rosbag2_test_common/include/rosbag2_test_common/publication_manager.hpp#L98-L152)
> 
> However usage of the `wait_for_matched` API for Player class via ros launcher might require to pass many parameters like number of matching nodes per topic and timeout. Maybe need to consider to pass them via some config file. I would assume that `wait_for_matched` would fully replace `delay` feature.

Does "blind delay" here mean users usually don't know how much time needed to wait?