ros2/rosbag2 #787: ros2 bag record exit code seems to be 2 when there's no error

russkel, 2021-06-22:
## Description
I noticed at the exit end of my launch logs any rosbag2 recording sessions would report an ERROR. It looks like `ros2 bag record` returns 2 even if it was successful. 

```
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rosbag-3] [INFO] [1624258420.697845659] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [rosbag-3]: process has died [pid 26949, exit code 2, cmd 'ros2 bag record -a -o /ros2_ws/rosbag'].
```

### Expected Behavior
Exit code of ros2 bag record command should be 0.

### Actual Behavior
```
ros@russ:/ros2_ws$ ros2 bag record -a -o rosbag2
[INFO] [1624329432.042607717] [rosbag2_storage]: Opened database '/ros2_ws/rosbag2/rosbag2_0.db3' for READ_WRITE.
[INFO] [1624329432.042690322] [rosbag2_recorder]: Listening for topics...
[INFO] [1624329432.043308341] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1624329432.043664406] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
^C[INFO] [1624329435.136884441] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1624329435.155487408] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
ros@russ:/ros2_ws$ echo $?
2
```

## To Reproduce
As per Actual Behaviour

## System (please complete the following information)
 - OS: focal
 - ROS 2 Distro: galactic
 - Version: ros-galactic-ros2bag/focal,now 0.9.0-1focal.20210521.032102 amd64 [installed,automatic]

emersonknapp, 2021-06-24:
Fixed in #788

closed 2021-06-24 by emersonknapp

russkel, 2021-06-24:
@emersonknapp, scholar and a gentleman. Thanks.