ros2/rosbag2 #790: Flaky test in test_rosbag2_record_end_to_end

Barry-Xu-2018, 2021-06-24:
## Description
test cases for test_rosbag2_record_end_to_end sometimes fails

### Expected Behavior
Get stabilization result

### Actual Behavior
Sometimes failure

## To Reproduce
Repeat run test_rosbag2_record_end_to_end in rosbags_test

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: rolling
 - Version: master

## Additional context
This problem also occurs in CI system

Barry-Xu-2018, 2021-06-24:
This problem was found at https://github.com/ros2/rosbag2/pull/788#issuecomment-866715732.  
And then I can reproduce this in my local environment if execute enough times.

Barry-Xu-2018, 2021-06-24:
I have analyzed problems at https://github.com/ros2/rosbag2/pull/788#issuecomment-866715732, but the solution need to be discussed here.  
So I copy my analysis to here.

---
I find CI report failure.  
> Rpr__rosbag2__ubuntu_focal_amd64  --- failure
According to https://build.ros2.org/job/Rpr__rosbag2__ubuntu_focal_amd64/881/testReport/, 3 test case failed.  

The failure reason includes 2 points. 
1. While publishing the message, sometimes rosbag doesn't discover the topic to be recorded. 
e.g. 
https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L121-L129

This leads to the error as the below (The number of publishing message doesn't match the number of recorded message)
https://build.ros2.org/job/Rpr__rosbag2__ubuntu_focal_amd64/881/testReport/rosbag2_tests/RecordFixture/record_end_to_end_test_with_cache/

Now we have no interface to make sure rosbag finish discovering topic, so I want to add sleep before publishing message to make sure discover is finished.

2. It is possible that  2 processes access the created sqlite3 database at the same time. 
https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L125-L127

start_execution() will start rosbag command in new process.  
https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_tests/test/rosbag2_tests/record_fixture.hpp#L120-L131
As the above codes, current process also access sqlite3 database.   
This leads that the lock occurs. 
```
[ RUN      ] RecordFixture.rosbag2_record_and_play_multiple_topics_with_filter
[INFO] [1624441972.716871010] [rosbag2_storage]: Opened database '/tmp/tmp_test_dir_OVIieg/rosbag2_record_and_play_multiple_topics_with_filter/rosbag2_record_and_play_multiple_topics_with_filter_0.db3' for READ_WRITE.
[INFO] [1624441972.716994585] [rosbag2_recorder]: Listening for topics...
[INFO] [1624441972.718430617] [rosbag2_recorder]: Subscribed to topic '/test_topic1'
[INFO] [1624441972.719189448] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
Error when processing SQL statement. SQLite error (5): database is locked
```

Barry-Xu-2018, 2021-06-24:
### Problem 1. While publishing the message, sometimes rosbag doesn't discover the topic to be recorded

This leads that first message is lost (Not record it) 
https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L121-L129

But I have no idea on how to confirm rosbag find all topics which will be recorded before `pub_manager.run_publishers()`  
Anybody has an idea ?

### Problem 2. It is possible that 2 processes access the created sqlite3 database at the same time.

https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_tests/test/rosbag2_tests/record_fixture.hpp#L120-L131

I think we need not open this file as sqlite3 data file to check. Instead, only check if file exists.

MichaelOrlov, 2021-06-26:
@Barry-Xu-2018 For the Problem 1 related to the discovery it would be more appropriate to add `wait_for_matched(..)` API for the PublicationManager. Similar as made [spin_and_wait_for_matched](https://github.com/ros2/rosbag2/blob/6f7503eafede497bda4f04c8e4b2de798b9cca99/rosbag2_test_common/include/rosbag2_test_common/subscription_manager.hpp#L82-L118) in SubscriptionManager.

I can do a PR if you are not sure how to do it, please let me know.

As regards to the `Problem 2`. It might be a race condition and fail on further stages if we will not wait while DB will be created. It could be situation when path exist but DB hasn't be created yet. If we will proceed we have a high chances to fail further on, when we will be doing request to the DB. Need to find a way how to check DB existence without locking access to it. May be some specific flags exists.
However I think the `Problem 2` will go away automatically as soon as we will solve `Problem 1` with wait for matched. Since first DB will be created in constructor of the recorder and than will be created subscribers and we will not try open DB until it will be actually created since we will be waiting for subscribers in test.

Barry-Xu-2018, 2021-06-27:
@MichaelOrlov  

> For the Problem 1 related to the discovery it would be more appropriate to add wait_for_matched(..) API for the PublicationManager. Similar as made spin_and_wait_for_matched in SubscriptionManager.  

Thanks. I will add this function and update test codes.   

> Need to find a way how to check DB existence without locking access to it. May be some specific flags exists.
>
> However I think the Problem 2 will go away automatically as soon as we will solve [...]

Okay. I will confirm this. If also have problems,  we can directly use sqlite3 API to check if this new DB file can be opened.

Barry-Xu-2018, 2021-06-28:
@MichaelOrlov 

Please help to review #797

Barry-Xu-2018, 2021-06-30:
Fix #797 was merged.  
So I close this issue.

closed 2021-06-30 by Barry-Xu-2018