ros2/rosbag2 #793: Record bag files with appropriate timestamp

k0suke-murakami, 2021-06-25:
## Description

Record rosbag files associated with the beginning time of the recording.

---
For example in ROS1, the recorded fils are with the time beginning of each file`s recording.

`$ roscore`
`$ rostopic pub -r 1.0 /test std_msgs/Bool true`
`$ rosbag record -a -b 1`

Saved files:
> 2021-06-25-15-00-28.bag
> 2021-06-25-15-00-54.bag
> 2021-06-25-15-01-11.bag
---
But in ROS2, multiple recorded files are not with the time beginning of each file`s recording, 
but with the initial time of the recording + index.

`$ ros2 topic pub /test std_msgs/msg/Bool data:\ true`
`$ ros2 bag record -a -d 1`
Saved files: 
>rosbag2_2021_06_25-15_10_54/
>├── metadata.yaml
>├── rosbag2_2021_06_25-15_10_54_0.db3
>├── rosbag2_2021_06_25-15_10_54_1.db3
>├── rosbag2_2021_06_25-15_10_54_2.db3
>├── rosbag2_2021_06_25-15_10_54_3.db3
>├── rosbag2_2021_06_25-15_10_54_4.db3
>├── rosbag2_2021_06_25-15_10_54_5.db3
>└── rosbag2_2021_06_25-15_10_54_6.db3

## Related Issues

Nothing as far as I see.

## Completion Criteria

## Implementation Notes / Suggestions
Have no idea how it can be implemented.

## Testing Notes / Suggestions

chaoflow, 2021-06-25:
@k0suke-murakami With rosbag1 I found it painful, that the files had different basenames, with rosbag2 the files are listed in ``metadata.yaml`` so I think the naming it not so important. What are your use cases?

I think it would make sense if ``metadata.yaml`` contained more per-file information, like start_time, duration, ... probably everything that is already available across all files.

k0suke-murakami, 2021-06-25:
> What are your use cases?
Suppose we have many chunked rosbag files from 8:00, 
and a bug reporter writes a report that having a bug at 9:32.

All we know from bag name is the initial time 8:00, it takes some time to search for the rosbag at 9:32.
But if we have rosbag file with the appropriate time, we can easily spot the rosbag file with the bug.

emersonknapp, 2021-06-28:
I am inclined to close this as something we will not implement. The timestamp you are seeing is only the default naming scheme, you can do e.g. `ros2 bag record -o mybag`, which will create

```
mybag/
├── metadata.yaml
├── mybag_0.db3
├── mybag_1.db3
├── mybag_2.db3
└── mybag_3.db3
```

The timestamp that you see is the "name of the bag". The "bag" is a directory, a single `db3` file is not a "bag", it is a storage backend file within a bag.

Ideally, you do not need to open individual files, instead you use a rosbag2 Read API to open a whole bag, which can open files as necessary. Of course, not all the functionality you need exists yet - but I think that's the direction we should go with implementation, rather than changing the naming scheme.

To start most basically, you can easily parse the metadata file to find the correct storage file in the bag.

That all being said, we could perhaps provide an option to suffix the individual files with a timestamp as an option - there is no required naming scheme for the files since the metadata lists them in full. However, certain tools like `ros2 bag reindex` do assume that there is a sequential order of `_N` (_0, _1, _2, ...) as the final part of the name.

WJaworskiRobotec, 2021-08-02:
@emersonknapp @chaoflow We discussed the solution with @k0suke-murakami and we would like to add more per file details to metadata.yaml (initially start_time and duration should be enough). What is your opinion about adding such extension to rosbag2 metadata file ?

emersonknapp, 2021-08-03:
Yes, it seems sensible to store per-file information in `metadata.yaml`

MichaelOrlov, 2024-05-05:
- Closing as per file information has been added to the metadata.

closed 2024-05-05 by MichaelOrlov