ros2/rosbag2 #794: relative_file_paths issue when replaying compressed bag file

nedo99, 2021-06-25:
## Description
When replaying the bag file, recorded with compression mode zstd, one must hardcode folder path plus file under `relative_file_paths` parameter in `metadata.yaml`.

### Expected Behavior
The `relative_file_paths` parameter should contain only the name of the bag file by default, the same as when one records the bag file in normal mode. ROS bag playing behavior should be the same for both compressed mode and non-compressed mode.

### Actual Behavior
When recording with compression mode, `relative_file_paths` by default contains the folder name plus the file name (e.g. `rosbag2_2021_06_25-11_59_52_0/rosbag2_2021_06_25-11_59_52_0.db3.zstd`). If we rename the recorded folder, playing of the bag file will not happen as the folder name is hardcoded (one will get errno(2)). This limits execution folder location since every time `metadata.yaml` file needs to be updated.

## To Reproduce
1. Record a bag file: `ros2 bag record --compression-mode file --compression-format zstd /ros/topic`
2. Rename recorded folder or execute `ros2 bag play <path/to/bag/folder>` from non-relative folder
3. You will get message
```
[ERROR] [1624628638.536812180] [rosbag2_transport]: Failed to play: Failed to open file: "<path>/rosbag2_2021_06_25-11_59_52_0.db3.zstd" for binary reading! errno(2)
```

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: 0.3.7-1focal.20210415.042502 (ros-foxy-ros2bag)

llapx, 2021-09-07:
It works well on rolling and galactic, maybe fixed in somewhere.

DLu, 2022-11-18:
Note: This is fixed in `foxy-future`, `galactic`, `humble` and `rolling` since https://github.com/ros2/rosbag2/pull/612

I opened #1161 to start to address the issue in `foxy`, and I have a fix for it, but I'm not sure what the intent is w.r.t. `foxy-future`. Should I still make the fix for this in a non-api breaking way?

MichaelOrlov, 2024-05-05:
- Closing as fixed in the #345

closed 2024-05-05 by MichaelOrlov