ros2/rosbag2 #802: Overriding QoS Profiles Doesn't Work

Shiyu-Liu, 2021-06-30:
## Description
** rosbag2 recording provides the possibility to override the qos profiles, in order to avoid the compatibility issue when recording the messages. However, this overriding feature is not working as expected.**

### Expected Behavior
** The ros2 bag record command should work with non default qos profiles. **

### Actual Behavior
** Even with the well defined qos overriding profile, the ros2 bag record still get the warning on the different qos profiles and no message is recorded. 
**

## To Reproduce
1. Publish msgs with best effort reliability on the topic _/talker_: 
    `ros2 topic pub -r 0.1 --qos-reliability best_effort /talker std_msgs/String "data: Hello World"`
2. Define the qos overriding profile with reliability as best effort in a YAML file (reliability_override.yaml): 
    `/talker:`
        `reliability: best_effort`
        `history: keep_all`
3. Record a ros2bag: `ros2 bag record -o my_bag /talker --qos-profile-overrides-path reliability_override.yaml`
4. Logs print error:
    `A new publisher for subscribed topic /talker was found offering RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT, but rosbag already subscribed requesting RMW_QOS_POLICY_RELIABILITY_RELIABLE. Messages from this new publisher will not be recorded.`

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: galactic
 - Version: (Installed by default with ros)

## Additional context
** The procedure is well followed and is referred to [Overriding QoS Policies For Recording And Playback](http://docs.ros.org.ros.informatik.uni-freiburg.de/en/crystal/Tutorials/Ros2bag/Overriding-QoS-Policies-For-Recording-And-Playback.html) **

emersonknapp, 2021-06-30:
This is probably an instance of https://github.com/ros2/rosbag2/issues/772 (the QoS isn't actually incompatible, but a warning is printed because of garbage values being checked)

1. Are you getting data in the rosbag? I expect that the topic is actually recorded, but that the warning is being printed incorrectly.
2. Can you test with the branch from https://github.com/ros2/rosbag2/pull/803 to see if this works for you?

And a note: `rosbag2` automatically chooses a compatible QoS for topics that exist when recording starts - you don't need to use QoS overrides unless you are in this specific situation:
* There are multiple publishers, on the same topic, offering _different_ QoS Profiles
* **AND** - you want to use a more restrictive QoS for recording, which would emit some of the publishers

Shiyu-Liu, 2021-07-01:
> This is probably an instance of #772 (the QoS isn't actually incompatible, but a warning is printed because of garbage values being checked)
> 
> 1. Are you getting data in the rosbag? I expect that the topic is actually recorded, but that the warning is being printed incorrectly (check contents of `bag/metadata.yaml` to see message counts easily)
> 2. Can you test with the branch from [Copy recorder QoS profile to local variable so that temporary value isn't cleared #803](https://github.com/ros2/rosbag2/pull/803) to see if this works for you?
> 
> And a note: `rosbag2` automatically chooses a compatible QoS for topics that exist when recording starts - you don't need to use QoS overrides unless you are in this specific situation:
> 
> * There are multiple publishers, on the same topic, offering _different_ QoS Profiles
> * **AND** - you want to use a more restrictive QoS for recording, which would emit some of the publishers

Thanks for your reply. 
Yes, what you said is true. I can get the data even with the warning printed out. However, I found that the ros2 bag record must be launched after the publisher being set up. 
In some cases, the publisher is not created at the beginning but only after something else has been triggered, while the subscriber node has been created from the beginning. And in this case, the rosbag recording is really timing-dependent. Meaning that if I record the bag along with the node, either written in the launch file or record from command line at the time when the programme starts, no message will be actually recorded after the actual messages are getting published. The warning shows that:
`New publisher discovered on topic '/topic_name', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY`

For now, I can just avoid this issue by recording manually after the publisher has started. But I would like to know if there are better solutions. Thanks!

emersonknapp, 2021-07-02:
> For now, I can just avoid this issue by recording manually after the publisher has started. But I would like to know if there are better solutions. 

Do you have multiple publishers on `/talker`? Rosbag2 will wait until your publishers are present - are you sure you're not getting any messages recorded?

closed 2021-07-02 by emersonknapp

Shiyu-Liu, 2021-07-02:
> Do you have multiple publishers on `/talker`? Rosbag2 will wait until your publishers are present - are you sure you're not getting any messages recorded?

It's a more complicated case than publishing to the topic `/talker`. But when I got the warning message `New publisher discovered on topic '/topic_name', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY`, it's indeed no message being recorded. 
But I guess I can modify my publisher node to solve it, like declaring the publisher when the node is initialized. 
Thanks for the reply anyway.

PR #803 (2021-07-02): Copy recorder QoS profile to local variable so that temporary value isn't cleared
Fixes #772
Fixes #802

This function was using a temporary variable that got cleared, and so had undefined behavior. Create a local variable to make the reference valid.