ros2/rosbag2 #820: [galactic] Cannot build from source as it includes the installed headers

kosuke55, 2021-07-13:
## Description
If rosbag2 is installed with apt, we cannot build from source.
This happens because PR https://github.com/ros2/rosbag2/pull/812  has been merged but has not been released on debian.
Wrong include directories order https://github.com/ros2/rosbag2/issues/698 may be helpful for include order.

### Expected Behavior
The build succeeds even if rosbag2 is installed.

### Actual Behavior
If rosbag2 is installed with apt, we cannot build from source.

## To Reproduce
clone and install rosbag2 package, build fails.
```
~/dev
$ git clone https://github.com/ros2/rosbag2.git -b galactic

~/dev
$ cd rosbag2/

~/dev/rosbag2 (galactic %=)
$ sudo apt install ros-galactic-rosbag2*

~/dev/rosbag2 (galactic %=)
$ colcon build --packages-up-to rosbag2_py
Starting >>> rosbag2_test_common
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_interfaces
Finished <<< sqlite3_vendor [0.22s]                                        
Finished <<< rosbag2_test_common [0.25s]
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [0.25s]
Finished <<< rosbag2_storage [0.24s]                                         
Starting >>> rosbag2_storage_default_plugins
Finished <<< rosbag2_interfaces [0.61s]                                   
Finished <<< rosbag2_storage_default_plugins [0.16s]
Starting >>> rosbag2_cpp
Finished <<< rosbag2_cpp [0.26s]                      
Starting >>> rosbag2_compression
Finished <<< rosbag2_compression [0.20s]                     
Starting >>> rosbag2_transport
Finished <<< rosbag2_transport [0.40s]                      
Starting >>> rosbag2_py
--- stderr: rosbag2_py                              
/home/kosuke55/dev/rosbag2/rosbag2_py/src/rosbag2_py/_transport.cpp: In function ‘void pybind11_init__transport(pybind11::module&)’:
/home/kosuke55/dev/rosbag2/rosbag2_py/src/rosbag2_py/_transport.cpp:228:41: error: ‘delay’ is not a member of ‘{anonymous}::PlayOptions’ {aka ‘{anonymous}::OptionsWrapper<rosbag2_transport::PlayOptions>’}
  228 |   .def_readwrite("delay", &PlayOptions::delay)
      |                                         ^~~~~
make[2]: *** [CMakeFiles/_transport.dir/build.make:63: CMakeFiles/_transport.dir/src/rosbag2_py/_transport.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:280: CMakeFiles/_transport.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_py [4.85s, exited with code 2]

Summary: 9 packages finished [6.58s]
  1 package failed: rosbag2_py
  1 package had stderr output: rosbag2_py
```

After removing rosbag2 packages, build can succeed.
```
~/dev/rosbag2 (galactic %=)
$ sudo apt remove ros-galactic-rosbag2* --purge

~/dev/rosbag2 (galactic %=)
$ colcon build --packages-up-to rosbag2_py
Starting >>> rosbag2_test_common
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_interfaces
Finished <<< rosbag2_test_common [0.15s]                                   
Starting >>> rosbag2_storage
Finished <<< sqlite3_vendor [0.16s]
Finished <<< shared_queues_vendor [0.18s]
Finished <<< rosbag2_storage [0.24s]                                         
Starting >>> rosbag2_storage_default_plugins
Finished <<< rosbag2_interfaces [0.52s]                                   
Finished <<< rosbag2_storage_default_plugins [0.16s]
Starting >>> rosbag2_cpp
Finished <<< rosbag2_cpp [2.98s]                      
Starting >>> rosbag2_compression
Finished <<< rosbag2_compression [1.63s]                      
Starting >>> rosbag2_transport
[Processing: rosbag2_transport]                               
[Processing: rosbag2_transport]                                      
Finished <<< rosbag2_transport [1min 4s]                              
Starting >>> rosbag2_py
Finished <<< rosbag2_py [11.7s]                             

Summary: 10 packages finished [1min 21s]
```

## System (please complete the following information)
 - OS: ubuntu20.04
 - ROS 2 Distro: galactic
 - Version: galactic

## Additional context
** Add any other context about the problem here **

emersonknapp, 2021-07-15:
Note: This is a symptom of https://github.com/ros2/ros2/issues/1150, and seems to be a duplicate of #698 - @hexonxons would you agree this is a duplicate issue?

hexonxons, 2021-07-15:
@emersonknapp yeah, seems like the same issue. I haven't merged fixes to master branch, just to foxy-future, so all other branches are still affected by this issue.

MichaelOrlov, 2024-05-05:
- Closing this issue since Galactic reached its End Of Life.

closed 2024-05-05 by MichaelOrlov