ros2/rosbag2 #823: ROS2 Bag Record Exclude does Not Work with All (-a)

tawheeler, 2021-07-21:
## Description
** clear and concise description of the bug is **
I was very happy to see that `ros2 bag record ` now support inclusion and exclusion regexes! 
I noticed, however, that exclusion does not play with with `-a`, and `-e .*` is required to get the desired behavior.

### Expected Behavior
** clear and concise description of what you _expected_ to happen **
I expect `ros2 bag record -a -x REGEX` to include all topics that do not match with the given regex REGEX, in the same way that `ros2 bag record -e '.*' -x  REGEX` behaves.

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

In my first terminal, I ran:
```
ros2 topic pub /potato std_msgs/msg/String "{data: 'hello'}"
```
At this point, `ros2 topic list` gives me:
```
/parameter_events
/potato
/rosout
```

If I run `ros2 bag record -e '.*' -x /pota.+` I get:
```
ros2 bag record -e '.*' -x /pota.+
1626833773.603844 [0]       ros2: using network interface wlp0s20f3 (udp/10.0.0.7) selected arbitrarily from: wlp0s20f3, docker0, tun0
[INFO] [1626833773.607125564] [rosbag2_storage]: Opened database 'rosbag2_2021_07_20-19_16_13/rosbag2_2021_07_20-19_16_13_0.db3' for READ_WRITE.
[INFO] [1626833773.607182830] [rosbag2_recorder]: Listening for topics...
[INFO] [1626833773.607892326] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1626833773.608474723] [rosbag2_recorder]: Subscribed to topic '/rosout'
[WARN] [1626833773.608588049] [rosbag2_recorder]: A new publisher for susbcribed topic /parameter_events was found offering RMW_QOS_POLICY_DURABILITY_VOLATILE, but rosbag2 already subscribed requesting RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL. Messages from this new publisher will not be recorded.
[WARN] [1626833773.608623901] [rosbag2_recorder]: A new publisher for susbcribed topic /parameter_events was found offering RMW_QOS_POLICY_DURABILITY_VOLATILE, but rosbag2 already subscribed requesting RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL. Messages from this new publisher will not be recorded.
^C[INFO] [1626833775.253514727] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1626833775.316412758] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
```
which recorded `/parameter_events` and `/rosout` as expected, ignoring `/potato`.

If I run `ros2 bag record -a -x /pota.+` I get:
```
ros2 bag record -a -x '/pota.+'
1626833846.270179 [0]       ros2: using network interface wlp0s20f3 (udp/10.0.0.7) selected arbitrarily from: wlp0s20f3, docker0, tun0
[INFO] [1626833846.273040991] [rosbag2_storage]: Opened database 'rosbag2_2021_07_20-19_17_26/rosbag2_2021_07_20-19_17_26_0.db3' for READ_WRITE.
[INFO] [1626833846.273088839] [rosbag2_recorder]: Listening for topics...
[INFO] [1626833846.273773987] [rosbag2_recorder]: Subscribed to topic '/rosout'
^C[INFO] [1626833848.736373393] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1626833848.785056772] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
```
which only recorded `/rosout`.
Running `ros2 bag record -a -x '/pota.+'` produces the same result.

## System (please complete the following information)
 - OS: Ubuntu 20.04.2 LTS
 - ROS 2 Distro: Galactic
 - Version: ros-galactic-desktop 0.9.3-2focal.20210521.072032

camm73, 2021-07-28:
This issue is now resolved via https://github.com/ros2/rosbag2/pull/825 and can be closed.

closed 2021-07-28 by tawheeler

emersonknapp, 2021-07-28:
Note that this fix won't be seen immediately, but will go out in the next Galactic sync.

tonynajjar, 2022-09-19:
> Note that this fix won't be seen immediately, but will go out in the next Galactic sync.

I think it's still not in galactic. Was rosbag2_transport released since then? I don't see a tag.

[image]