ros2/rosbag2 #828: test_play_next failed on nightly CI with an assertion

tfoote, 2021-07-30:
## Description

rosbag2_transport.test_play_next__rmw_fastrtps_cpp.gtest.missing_result test failures

### Expected Behavior
Pass the test

### Actual Behavior
The test failed with the following assertion: 

```
[       OK ] RosBag2PlayTestFixture.play_next_playing_all_messages_without_delays (168 ms)
[ RUN      ] RosBag2PlayTestFixture.play_next_playing_one_by_one_messages_with_the_same_timestamp
test_play_next__rmw_fastrtps_cpp: /home/jenkins-agent/workspace/nightly_linux_debug/ws/install/shared_queues_vendor/include/moodycamel/readerwriterqueue.h:646: moodycamel::ReaderWriterQueue<T, MAX_BLOCK_SIZE>::ReentrantGuard::ReentrantGuard(bool&) [with T = std::shared_ptr<rosbag2_storage::SerializedBagMessage>; long unsigned int MAX_BLOCK_SIZE = 512]: Assertion `!inSection && "ReaderWriterQueue does not support enqueuing or dequeuing elements from other elements' ctors and dtors"' failed.
```

https://ci.ros2.org/job/nightly_linux_debug/2005/testReport/junit/(root)/rosbag2_transport/test_play_next__rmw_fastrtps_cpp_gtest_missing_result/

## To Reproduce
Nightly linux debug CI: https://ci.ros2.org/job/nightly_linux_debug/

## System (please complete the following information)
 - OS: Focal
 - ROS 2 Distro: rolling
 - Version: master

tfoote, 2021-07-30:
There appear to have been several instances of this failure last night this is likely more than just flaky.

* http://build.ros2.org/job/Rci__nightly-debug_ubuntu_focal_amd64/395/testReport/
* http://build.ros2.org/job/Rci__nightly-fastrtps-dynamic_ubuntu_focal_amd64/395/testReport/
* https://ci.ros2.org/job/nightly_linux_debug/2005/testReport/

MichaelOrlov, 2021-07-30:
Preliminary analysis:
The moodycamel::ReaderWriterQueue triggering assert in
```
struct ReentrantGuard
	{
		ReentrantGuard(bool& _inSection)
			: inSection(_inSection)
		{
			assert(!inSection && "ReaderWriterQueue does not support enqueuing or dequeuing elements from other elements' ctors and dtors");
			inSection = true;
		}

		~ReentrantGuard() { inSection = false; }

	private:
		ReentrantGuard& operator=(ReentrantGuard const&);

	private:
		bool& inSection;
	};
```
When calling `peek()` method from two threads simultaneously
```cpp
	T* peek()
	{
#ifndef NDEBUG
		ReentrantGuard guard(this->dequeuing);
#endif
```
The situation when we are calling `peek()` from two threads happening when we are starting playback in one thread and right after that calling `Player::play_next()` API.
The first call for `peek` happening here https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/player.cpp#L377 and the second call for the `peek` from `Player::play_next()` API is happening here https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/player.cpp#L321

When I was writing synchronization for `play_next` and `play` API I used assumption that it's safe to call `moodycamel::ReaderWriterQueue::peek()` API from two separate threads simultaneously and it's actually safe. 
This is just falsely triggering assertion inside `moodycamel::ReaderWriterQueue::peek()` method. 
By design `moodycamel::ReaderWriterQueue` declared as **Single producer - Single consumer** lock free queue. And `ReentrantGuard` is intend to assert when someone trying to `dequeue` from more than one thread. But in our case we are actually not trying to dequeue element concurrently rather just taking pointer to the top element in the queue concurrently which is legitimate operation and doesn't affect data integrity since this is just read only operation.

As a workaround on our "Rosbag2" side I can propose to move section 
```cpp
 {  // Notify play_next() that we are ready for playback
        std::lock_guard<std::mutex> lk(ready_to_play_from_queue_mutex_);
        is_ready_to_play_from_queue_ = true;
        ready_to_play_from_queue_cv_.notify_all();
  }
```
from [void Player::play()](https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/player.cpp#L239-L243)
right after
```cpp
  rosbag2_storage::SerializedBagMessageSharedPtr * message_ptr = peek_next_message_from_queue();
```
in [void Player::play_messages_from_queue()](https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/player.cpp#L377)

@tfoote @emersonknapp If you are agree with my analysis and proposed workaround I can implement fix fro this failure.

PR #832 (2021-07-30): Move notification about ready for playback inside play_messages_from_…
Move notification that player ready for playback inside play_messages_from_queue() after peeking pointer to the next message.
The rational is that `moodycamel::ReaderWriterQueue::peek()` is not allowed to be called from more than one thread concurrently. 
Please see more details in https://github.com/ros2/rosbag2/issues/828#issuecomment-889700778

Option with protecting `Player::peek_next_message_from_queue()` by mutex is highly undesirable since it would affect performance in case we would have messages recorded with high  [... cut]
closed 2021-07-30 by MichaelOrlov