ros2/rosbag2 #829: Discovery silently stops after unknown msg type is found

alsora, 2021-07-30:
## Description
I'm running `ros2 bag record --all`.
This should keep trying discovering topics and subscribe to them once they are found.
This works as expected if all the topics present on the network have known message types, but it silently fails when a topic with unknown message type is discovered.

### Expected Behavior
Obviously we can't currently subscribe to topics with unknown message types, so, when they are discovered, a warning message should be displayed by rosbag and they should simply be ignored, allowing everything else to continue.

### Actual Behavior
As soon as a topic with unknown message type is discovered, the rosbag process stops discovering new topics even if they have known message types and does not show any hint to the user about what is going on.

## To Reproduce
 - Have 2 different pacakges with ROS messages, A and B, source only package A in the rosbag shell.
 - Start rosbag with `ros2 bag record --all`
 - Start publishing on 1st topic with a message type A (this type is known to rosbag and rosbag will log that it subscribed to it)
 - Start publishing on 2nd topic with a message type B (this type is not known to rosbag and rosbag will not subscribe to it, but nothing will be logged)
 - Start publishing on a 3rd topic with message type A (this type is known to rosbag, but it will not be logged due to the bug from this ticket.

After the experiment, it's possible to inspect the bag with `ros2 bag info` and see that it only contains the 1st topic but not the 3rd.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic

emersonknapp, 2021-08-03:
Good bug report - thanks for the detailed information.

emersonknapp, 2021-08-30:
Fixed in #848
Galactic backport in #858

closed 2021-08-30 by emersonknapp