ros2/rosbag2 #830: Confusing warning message about incompatible QoS

alsora, 2021-07-30:
## Description
I'm publishing on a topic with a BestEffort publisher and I'm trying to record it using `ros2 bag record --all`.
ROSbag prints a very confusing warning.

### Expected Behavior
ROSbag subscribes to the topic and records the messages.

### Actual Behavior
I get the following warning log
```
[WARN] [1627639697.480796247] [rosbag2_recorder]: A new publisher for subscribed topic /wheel_vels was found offering RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT, but rosbag already subscribed requesting RMW_QOS_POLICY_RELIABILITY_RELIABLE. Messages from this new publisher will not be recorded.
```
This seems to indicate that something is wrong with my system and/or messages will not be recorded.
However, everything still works and I can find messages in my rosbag.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic

clalancette, 2021-08-02:
I think this is a duplicate of #772, and should be fixed in the next Galactic sync.  But I'll leave it to @emersonknapp to confirm and close.

emersonknapp, 2021-08-03:
Yes, this was fixed in https://github.com/ros2/rosbag2/pull/803. This should be solved in the next Galactic sync. IMO it should have been in the last sync, I had the rosdistro PR there on what I understood to be the due day - but it didn't go in, so I was just confused - the messaging around dates could be more consistent and precise.

KhalidOwlWalid, 2022-01-16:
I was just wondering how I could solve this issue in rclpy? 

From what I understand, it was suggested for me to follow this [API Break with get_actual_qos()](https://github.com/ros2/ros2_documentation/blob/0aac77333922ac490d356125cb6de4ee07c8a207/source/Releases/Release-Eloquent-Elusor.rst#api-break-with-get_actual_qos) but I cant seem to find anything for rclpy

emersonknapp, 2022-02-03:
Closing as fixed - please reopen if needed

closed 2022-02-03 by emersonknapp