ros2/rosbag2 #834: Unable to install rosbag2 in windows

anacsousa1, 2021-08-03:
Hello! Any help is appreciated. Thank you!

## Description
I've been trying to install rosbag2 in windows, but my installation stops in zstd_vendor. It opens the patch in another window, but even after 2h, it does not finish the installation.

### Expected Behavior
I expected the installation to finalize successfully or stops and log the error.

### Actual Behavior

In cmd:
> C:\Users\anacs\Desktop\sandbox\rosbag2>colcon build --merge-install
> Starting >>> rosbag2_test_common
> Starting >>> shared_queues_vendor
> Starting >>> zstd_vendor
> Starting >>> sqlite3_vendor
> Finished <<< sqlite3_vendor [12.7s]
> Finished <<< rosbag2_test_common [15.0s]
> Starting >>> rosbag2_storage
> Finished <<< shared_queues_vendor [19.8s]
> [Processing: rosbag2_storage, zstd_vendor]
> --- stderr: rosbag2_storage

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Create a new directory: rosbag2/src
2. Cloned the foxy branch from rosbag2: `git clone --branch foxy https://github.com/ros2/rosbag2.git`
3. Build: `colcon build --merge-install`
4. There are no log print errors because it never finishes.

## System
 - OS: Windows 10
 - ROS 2 Distro: foxy
 - Version: branch foxy

## Additional context
I have a successful ros2 foxy build in my Win10 that works fine (I can develop and run most nodes without any problems).
I've been able to install rosbag2 for Ubuntu, but something is hanging in Windows...

clalancette, 2021-08-03:
Likely this is because you need to start your terminal in Administrator mode.  With that, I'm guessing this will work.

That said, this is a bug in the zstd_vendor package.  It should be switched to using `git apply` rather than `patch`, as that will work without Administrator.

emersonknapp, 2021-08-03:
related to #632

anacsousa1, 2021-08-04:
I started in Adm mode, and things happened. Thank you.
It compiled everything except **rosbag2_compression**, the log error is:

`c:\users\anacs\desktop\sandbox\rosbag2\src\rosbag2\rosbag2_compression\src\rosbag2_compression\compression_factory_impl.hpp(36): error C2039: 'tolower': is not a member of 'std' [C:\Users\anacs\Desktop\sandbox\rosbag2\build\rosbag2_compression\rosbag2_compression.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include\iostream(14): note: see declaration of 'std'`
  
  Yet, basic commands seem to be working.

closed 2021-08-04 by anacsousa1