ros2/rosbag2 #855: Ability to export ROS 2 bag

audrow, 2021-08-24:
## Description

Currently it seems hard to share ROS 2 bags. It would be nice if this was easier.

## Completion Criteria

There is a simple, unambiguous way to share ROS 2 bag files, that is to save the bag, send it to someone else or another computer, and then play the bag.

## Implementation Notes / Suggestions

It was suggested in the ROS 2 weekly meeting to compress the directory created by ROS 2 bag into, for example, a tar file. This could be done with an `--export` flag.

aprotyas, 2021-08-24:
Just poking: do you see the `--export` flag doing anything more than generating an archive of the directory produced by `ros2 bag record`?

audrow, 2021-08-24:
I'm not sure. The goal is to make it easier to share bag files and creating an archive was suggested. I'm not sure if that gets us all the way there, but probably not. I suspect that different configurations will pose a problem, but I'm not terribly knowledgable or experienced with ROS bag 2.

emersonknapp, 2021-08-24:
> It seems hard to share

We have decided that `tar` is hard to use? Not meaning to be argumentative, but that seems like a pretty normal thing to do to share files/directories out in the wild. Maybe we want rosbag2 to call `tar` for you?

danthony06, 2021-08-24:
Is this supposed to do anything other than compress the entire directory? I'm not seeing the challenge here either. If we do need this, maybe use zip instead of tar, just so Windows can process them with its built in tools?

Karsten1987, 2021-08-25:
there might be potentially more work to do than just compression the directory. The `metadata.yaml` file can have absolute filepaths pointing to files outside the current directory. I'd envision the logic to be a bit more complex in which one should parse the metadata, copy (?) the files which are not inside the directory into it and then compress.

audrow, 2021-08-25:
FYI @defunctzombie @jtbandes @amacneil @jhurliman

danthony06, 2021-08-30:
We talked about this quite a bit in the WG meeting, and for background, this i the workflow we're trying to support.

We currently maintain a utility called Bag Database (https://github.com/swri-robotics/bag-database). It's a web based utility that lets us keep track of and manage our bag files. As part of these, we read things like latitude and longitude out of GPSFix or NavSatFix messages so that users can see where the bags were recorded, the time they were recorded, etc. All of this is done using a pure Java implementation of a ROS 1 bag parser so we have no dependencies on actual ROS utilities. This is really desirable, because internally we have terabytes of bags spanning years of development efforts both from us and other partners, so it's important to us that the bag format be self describing so that we can introspect the bags and see what's in them, without needing the actual message definitions used to create that bag file.

It's also important that we are able to read chunks from the bag files, because now we can integrate with utilities like Webviz and Foxglove, and you can open the bags within the bag database and send the data over to these types of visualization and analysis tools.

amacneil, 2022-03-13:
I think we should close this ticket because it is not well defined. The main request is to make the file format self-contained, which was a primary design goal for [mcap](https://foxglove.dev/blog/introducing-the-mcap-file-format) and will be available for testing in ROS 2 (as an alternative to sqlite) very soon.

jhurliman, 2022-03-13:
The ROS2 MCAP storage plugin is ready for testing, please try it out and provide feedback: https://github.com/ros-tooling/rosbag2_storage_mcap

audrow, 2022-03-15:
> I think we should close this ticket because it is not well defined. The main request is to make the file format self-contained, which was a primary design goal for [...]

Sounds good. Closing.

closed 2022-03-15 by audrow