ros2/rosbag2 #856: Visiting invalid pointer addresss when using "MESSAGE" compression mode to write data

0xbabeface, 2021-08-25:
## Description
I wrote testing codes like: 
```cpp
rosbag2_compression::CompressionOptions compression_options{
    "zstd",
    rosbag2_compression::CompressionMode::MESSAGE,
    10,
    1,
};
writer_ = std::make_unique<rosbag2_cpp::Writer>(
    std::make_unique<rosbag2_compression::SequentialCompressionWriter>(compression_options));
// Do somthing to initialize the writer_, like getting the path and opening it with some options
for (;;) {
  writer_->write(msg_, topic_, rclcpp::Clock().now());
}
```
The program will throw out a "std::bad_alloc" exception when writing bag.

### Expected Behavior
Work normally.

### Actual Behavior
In the function in rosbag2_cpp/writer.cpp:
```cpp
void Writer::write(
  const rclcpp::SerializedMessage & message,
  const std::string & topic_name,
  const std::string & type_name,
  const rclcpp::Time & time)
{
  auto serialized_bag_message = std::make_shared<rosbag2_storage::SerializedBagMessage>();
  serialized_bag_message->topic_name = topic_name;
  serialized_bag_message->time_stamp = time.nanoseconds();
  
  // temporary store the payload in a shared_ptr.
  // add custom no-op deleter to avoid deep copying data.
  serialized_bag_message->serialized_data = std::shared_ptr<rcutils_uint8_array_t>(
    const_cast<rcutils_uint8_array_t *>(&message.get_rcl_serialized_message()),
    [](rcutils_uint8_array_t * /* data */) {});

  // ---
  // The write behaviour will result in `write` function in SequentialCompressionWriter:   writer_impl_->write(message);
  // And serialized_bag_message will be pushed in to a queue to zstd compressor in another thread
  // However, message in this thread may have been destructed when visiting serialized_bag_message.
  // So that the pointor `serialized_bag_message->serialized_data` is hanged 
  // and the compressor visits pointer with a invalid address
  // ---
  return write(
    serialized_bag_message, topic_name, type_name, rmw_get_serialization_format());
}

```

## To Reproduce
Maybe some codes for testing are needed.

## System (please complete the following information)
 - OS: focal
 - ROS 2 Distro: galactic
 - Version: release

## Additional context
I made a deepcopy from the input parameter `message` to a new piece of memory. Then the program works.
But I am not familiar with memory allocation in rclcpp and don't know if it is safe, so I will not make a PR to fix it.

Barry-Xu-2018, 2021-09-03:
Agree. This is an issue. 
Currently, only this interface has this problem. 
```c++
void Writer::write(
  const rclcpp::SerializedMessage & message,
  const std::string & topic_name,
  const std::string & type_name,
  const rclcpp::Time & time)
```
Fortunately, this interface isn't really used.

I think  
`const rclcpp::SerializedMessage & message` is unreasonable definition.   
`std::shared_ptr<rclcpp::SerializedMessage> message` should replace it.   
That is, interface is changed to 
```c++
void Writer::write(
  std::shared_ptr<rclcpp::SerializedMessage> message,
  const std::string & topic_name,
  const std::string & type_name,
  const rclcpp::Time & time)
```

Besides, below codes can be wrapped into above interface and simplify codes in recorder.  

https://github.com/ros2/rosbag2/blob/891e08128e2d6ff36452871abda2cd776b8a1566/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L223-L248

@emersonknapp @Karsten1987 what do you think ?

emersonknapp, 2021-09-03:
I think that change is fine - however it's now a fixed API in Galactic, so we wouldn't be able to backport it. We could make this change for Humble - but is there any patchfix we can do for Galactic without changing the API?

Barry-Xu-2018, 2021-09-06:
> I think that change is fine - however it's now a fixed API in Galactic, so we wouldn't be able to backport it. We could make this change for Humble - but is there any patchfix we can do for Galactic without changing the API?

Okay. I will update codes to fix this bug.  
And make another PR for new interface.
```
void Writer::write(
  std::shared_ptr<rclcpp::SerializedMessage> message,
  const std::string & topic_name,
  const std::string & type_name,
  const rclcpp::Time & time)
```

Barry-Xu-2018, 2021-09-07:
@emersonknapp 

I created 2 PRs to fix this problem.   
- #866 : always duplicate message
- #867 : Add new interface to help judgement whether processing message is asynchronous. Only for asynchronous, duplicate message

Would you please share your opinions on fix ?

MichaelOrlov, 2023-02-01:
- Closing as resolved in #866

closed 2023-02-01 by MichaelOrlov