ros2/rosbag2 #859: [CONSIDER] Add ros2 bag reindex to foxy-future branch

EsipovPA, 2021-08-31:
## Description

Currently, the reindexing feature is available in master branch. Also in the first iteration of ROS. Is it a viable option to consider adding the same functionality to the  foxy-future branch? I know that this branch is not going to last for long, but this feature might come in handy in some scenarios. At least, I've encountered one not long ago.

## Related Issues

Did not find any yet

## Completion Criteria

Duplicate of the master branch

 - implementation of reindexer class for foxy-future branch
 - 'reindex' verb for 'ros2 bag'
 - some python bindings maybe?

## Implementation Notes / Suggestions

Moving cpp and hpp files from master do not work ( didn't work for me at least, lol ), but with minor fixes, this might actually work. In my case, the QoS parameters gone missing in reindexed rosbag.

## Testing Notes / Suggestions

There is already a test for Reindexer class in master, so moving it to foxy-future doesn't look too complicated to me.

emersonknapp, 2021-08-31:
A couple nitpicks with the issue:
* this feature is available in Galactic and Rolling, not just from code build of master branch
* Backports are not implemented by copying files from one branch to another - they are done by cherry-picking specific commits/diffs so that only the changes relevant to that feature are actually picked.

Since the reindexing feature was implemented _directly_ after the cutoff for the foxy-future branch, the backport may not be too bad. I've put up https://github.com/ros2/rosbag2/pull/860/ to try it out, but I can't promise that it's a clean merge. The main concern with any backport is that it does not break any existing APIs (normally ABI is also a consideration but we don't have that promise for `foxy-future`).

I'm going to close this issue in favor of just tracking the backport work in the PR.

closed 2021-08-31 by emersonknapp

EsipovPA, 2021-09-01:
This is a very good news, that there is already a PR. Thank you very much!