ros2/rosbag2 #876: Add Writer.close() API

jacobperron, 2021-10-04:
## Description

The `BaseWriterInterface` declares a [`close()` method](https://github.com/ros2/rosbag2/blob/e6867e7a488815665112c556c92cf970bffa2464/rosbag2_cpp/include/rosbag2_cpp/writer_interfaces/base_writer_interface.hpp#L41) that all writer implementations must override. However, the top-level [`Writer` class](https://github.com/ros2/rosbag2/blob/e6867e7a488815665112c556c92cf970bffa2464/rosbag2_cpp/src/rosbag2_cpp/writer.cpp) does not call this anywhere. I'm not sure if this is by design or an oversight. It seems to me that it would be valuable to add an explicit close method to `Writer`

## Completion Criteria

* [ ] Add `close()` method to `Writer`
* [ ] Expose this method in `rosbag2_py`
* [ ] [Optional] Use in `rosbag2_py.Recorder` to improve usage

## Implementation Notes / Suggestions

Note, the `Reader` implements a similar `close()` method:

https://github.com/ros2/rosbag2/blob/44e6422f40d424ebf3e741af058192e0b93b925a/rosbag2_cpp/src/rosbag2_cpp/reader.cpp#L55-L58

## Testing Notes / Suggestions

If I'm not mistaken, adding `close()` method should make it easier to write some tests. For example, I ran into this issue trying to use the Python API; we could refactor the following test code:

[test_sequential_writer.py](https://github.com/ros2/rosbag2/blob/e6867e7a488815665112c556c92cf970bffa2464/rosbag2_py/test/test_sequential_writer.py#L74-L75)

```diff
    # close bag and create new storage instance
-    del writer
+   writer.close()
```

[test_recorder.py](https://github.com/ros2/rosbag2/blob/e6867e7a488815665112c556c92cf970bffa2464/rosbag2_py/test/test_transport.py#L82-L87)

```diff
    recorder.cancel()

    metadata_path = Path(bag_path) / 'metadata.yaml'
    db3_path = Path(bag_path) / 'test_record_cancel_0.db3'
-    assert wait_for(lambda: metadata_path.is_file() and db3_path.is_file(),
-                    timeout=rclpy.duration.Duration(seconds=3))
+    assert metadata_path.is_file()
+    assert db3_path.is_file()
```

adamconkey, 2023-01-18:
I'll note that `SequentialWriter` does in fact implement the `close` method [here](https://github.com/ros2/rosbag2/blob/e6867e7a488815665112c556c92cf970bffa2464/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp#L140-L155). 

However, I find this issue particularly relevant because it is not exposed in the Python API. Are there any plans for adding it to the Python API? For my purposes the Python API is useless without it, since I don't want to have to rely on my node being killed for the bag file to close out. I want to have control over that.

MichaelOrlov, 2023-01-19:
No dedicated plans. However as always PRs are welcome.

laurafbec, 2023-01-24:
I would also be interested in this enhacement. Just for the record, is there an estimated release date? Depending on it I would consider the python or c++ implementation. Thanks.

MichaelOrlov, 2023-06-01:
- Closing as done in #1339 and #1300

emersonknapp, 2023-06-07:
Looks like you missed the close button - please reopen if I am mistaken

closed 2023-06-07 by emersonknapp