ros2/rosbag2 #900: Snapshot based on duration

CourchesneA, 2021-11-01:
## Description

There was a recent addition of the snapshot feature that allows a circular buffer of fixed size. In our case, it is not possible to specify a buffer zise, but we would like to specify a duration. It would be great to be able to trigger a snapshot for X seconds in the past and Y seconds in the future.

## Related Issues

#663 

## Completion Criteria

- Able to trigger a snapshot based on time instead of memory

## Testing Notes / Suggestions

Extra goals would be support for multiple overlapping snapshots

AntoineHX, 2022-07-04:
Are their plans to integrate this to Foxy-future ?
I believe [https://github.com/gaia-platform/rosbag2_snapshot](https://github.com/gaia-platform/rosbag2_snapshot), now, has this feature. But it seems only compatible with Galactic or above.

MichaelOrlov, 2022-07-14:
@AntoineHX There are no plans or scheduled effort for backporting `rosbag2_snapshot` feature to the https://github.com/ros2/rosbag2/tree/foxy-future branch.  
As always PRs are welcome.

Stefan-working, 2022-11-18:
Are there any plans to implement this enhancement to the rosbag2? 
If I am reading the doc and source of the  https://github.com/gaia-platform/rosbag2_snapshot correctly this still misses the feature to also further record Y seconds into the future and rosbag2_snapshot seems to have no license.
I really would appreciate it if this feature would be part of the rosbag2 core, as we really like to use this feature for recording only specific scenarios and use cases.

CourchesneA, 2022-11-18:
@Stefan-working  In the end we ended up using the GAIA's implementation (which I believe is not unmaintained). We were able to implement "record Y seconds into the future" manually with a simple sleep(Y).

MichaelOrlov, 2022-11-19:
@Stefan-working As regards to the license. 
I found this line
https://github.com/gaia-platform/rosbag2_snapshot/blob/333ae57b9f606c3e69da7b968d0e8ba44ccc5374/rosbag2_snapshot/package.xml#L8

Stefan-working, 2022-11-21:
Thank you @CourchesneA and @MichaelOrlov with this license we may be able to use it and adapt it with the sleep.

tonynajjar, 2024-03-25:
Reviving this issue, this would be quite useful to me as well!

MichaelOrlov, 2024-03-26:
Sorry, don't have capacity for this feature for a while. It is supposed to be structural changes and not straightforward to implement.

tonynajjar, 2024-03-26:
> Sorry, don't have capacity for this feature for a while. It is supposed to be structural changes and not straightforward to implement.

It's the issue with the most "thumbs up" though, or how is work prioritized in this repo?

[image]

MichaelOrlov, 2024-04-14:
Priority is assigned according to the importance and feasibility.
Priority is low because there exists a workaround (one can estimate data stream size and specify buffer size in bytes) and also exists a third-party solution. Another reason why the priority is low is because potential implementation is not straightforward and requires structural changes in the core of the rosbag2 on multiple layers.

I personally was planning to get back to this issue/feature somewhere in the June-July when dust after the Jazzy release will settle down.