ros2/rosbag2 #901: Bagfile can't be played when recording is not stoped appropriately

fas2037, 2021-11-02:
## Description
I'm running ROS2 Galactic in an Docker Container on a Jetson Xavier NX. When starting the docker a rosbag is recorded and all nodes are started automatically. If I then run docker stop and try to run the bag file the following error occurs 
```
Could not open 'rosbag2_...' with 'sqlite3'. Error: Failed to setup storage. Error: Could not read-only open database. SQLite error (10): disk I/O error
Could not load/open plugin with storage id 'sqlite3'
```

When starting and stopping the ros2 bag record manually inside of the docker container it is possible to play the file afterwards.

### Expected Behavior
As in ros1 bags a corrupted bag file should be repaired and still playable

### Actual Behavior
Bag cannot be played

## System (please complete the following information)
 - Docker container with Ubuntu 20.04 and ROS2 Galactic on Jetson Xavier NX (arm64, Ubuntu 18.04)

fas2037, 2021-11-02:
When runing ros2 bag play with the database file (.db3) it works

closed 2021-11-02 by fas2037

emersonknapp, 2021-11-02:
This is happening because of an unclean shutdown, the process is being terminated without being allowed to do any finalizing work (which is where the metadata file is created). In your case, does the bag directory that you are trying to play back have `metadata.yaml` in it?

Although - my understanding is that a `docker stop` sends `SIGTERM` to the main process, and rosbag2 had a specific galactic backport to gracefully handle SIGTERM - https://github.com/ros2/rosbag2/pull/807

Could you do the following experiment?
* Run your `ros2 bag record` command
* Within the container, send a SIGTERM to the recording process (`kill -s SIGTERM <PID>`)
* See if you can play that bag back successfully from the top-level directory (also see if it has `metadata.yaml` in it)

If that case works OK - then something with the `docker stop` is not sending the right signals