ros2/rosbag2 #902: Composable `Player` and `Recorder` nodes

hidmic, 2021-11-02:
## Description

To enable intra-process bag replaying and recording, sometimes necessary when dealing with high-frequency and/or high-bandwidth data streams, both [`rosbag2_transport::Player`](https://github.com/ros2/rosbag2/blob/1f68ddce236acc9425ead2fd87bf5dbc9c9407d2/rosbag2_transport/include/rosbag2_transport/player.hpp#L59) and [`rosbag2_transport::Recorder`](https://github.com/ros2/rosbag2/blob/1f68ddce236acc9425ead2fd87bf5dbc9c9407d2/rosbag2_transport/include/rosbag2_transport/recorder.hpp#L46) nodes have to be exposed as `rclcpp` components.

## Completion Criteria

- [ ] Allow `rosbag2_transport::Player` node configuration via parameters.  
- [ ] Register `rosbag2_transport::Player` node as a component.
- [ ] Allow `rosbag2_transport::Recorder` node configuration via parameters.
- [ ] Register `rosbag2_transport::Recorder` node as a component.

## Testing Notes / Suggestions

Updating `test_record.cpp` and `test_play.cpp` to use intra-process comms and inter-process comms for its nodes should do.

berndpfrommer, 2022-02-22:
I just checked and as of today 2/22/2022 the modified header files have not made it into the official ubuntu galactic packages. Is there a schedule for when that will happen?

doisyg, 2022-11-13:
Also interested in having composable and IPC player capability. What is the current status for Humble ? Any plans ?

berndpfrommer, 2022-11-13:
Is introducing PR #892 an option? I don't think it would interfere with implementing a better solution later down the road and it would keep [my stopgap solution ](https://github.com/berndpfrommer/rosbag2_composable_recorder) alive.

roncapat, 2023-07-12:
Is someone already tackling this? Otherwise, I may try to open a PR as soon as I find some spare time

emersonknapp, 2023-07-12:
@roncapat - @Karsten1987 has moved on to other projects and hasn't been working on rosbag2 for a while, none of the active maintainers currently have this on their plate, so contribution is very welcome. Perhaps you'd like to attend the rosbag2 working group Friday morning to discuss the design? https://calendar.google.com/calendar/u/0/embed?src=agf3kajirket8khktupm9go748@group.calendar.google.com (note default UTC time zone). If that's not feasible for you we can discuss over GitHub - feel free to open draft PRs early in development and ping @MichaelOrlov and myself to talk about overall structure.

roncapat, 2023-07-12:
@emersonknapp thank you very much for feedback! I'll try (and let you know) if I can manage to align friday meeting in my agenda, otherwise I will surely draft something here on github and discuss with you here - happy to contribute if I can :)

roncapat, 2023-07-13:
So, let's start with the first issue: `RecordOptions`, `PlayOptions` and `StorageOptions` feature many `uint64_t` and even some `size_t` attributes.
However, rclcpp parameter API only supports signed integers, being `int64_t` the widest. We would loose the ability to set the upper half of the `uint64_t` value range.
About `size_t`, it is both unsigned and platform dependent (even if I think on 90% of current ROS 2 setups with 64 bit x86 or ARM, it should correspond again to an `uint64_t`).
Do you have an opinion on this, or experience from other packages? Can we afford to loose half of the positive interval for those values?

emersonknapp, 2023-07-21:
Personal opinion on that opinion - it ain't ugly if it's needed. We can just validate that the parameters are positive, and cast them to the necessary types. No big deal. Anybody who wants that extra range is doing something very special and can work around it by not using Parameters for configuration.

roncapat, 2023-07-21:
Thanks! BTW, the linked PR builds fine, but I will probably have time to test it properly not earlier than August. 

If you have time to review even part of the proposed code in the meantime, I will be very happy to incorporate feedbacks :)

nachovizzo, 2023-11-02:
Any update on this one folks :angel: ? I'm happy to jump and help if that's neccesary

roncapat, 2023-11-02:
> Any update on this one folks 👼 ? I'm happy to jump and help if that's neccesary

Working hard right now on this prerequisite #1476! Feel free to follow or suggest improvements of course. #1476 started as a small tweak to better support configurability of QoS from parameter files, and became quite a huge patch due to all the inter-dependencies here-and-there in the codebase.

roncapat, 2023-12-01:
@nachovizzo lots of updates on the associated PR. Would you like testing / provide some feedback?

nachovizzo, 2023-12-19:
> @nachovizzo lots of updates on the associated PR #1419. Would you like testing / provide some feedback?

Sorry, I missed this notification. Silly me. Feedback: Extremely happy to see this got merged :D 

Great job man!!!

roncapat, 2023-12-19:
No worries, still valid :) If you ever happen to leverage this feature and discover issues, tag me and let me know!

nachovizzo, 2023-12-19:
Sure thing ;)

PR #1419 (2023-12-11): Make Player and Recorder Composable (#902)
## Completion Criteria
 * [x]  Allow `rosbag2_transport::Player` node configuration via parameters.
 * [x]  Register `rosbag2_transport::Player` node as a component.
 * [x]  Allow `rosbag2_transport::Recorder` node configuration via parameters.
 * [x]  Register `rosbag2_transport::Recorder` node as a component.
 * [x]  Write tests
     * [x]  Be able to access to the protected storage_options, player_options and record_options for verifications via `mock_player` and `mock_recorder`.
     * [x]  Create node with expected parameters and check all parame [... cut]