ros2/rosbag2 #906: New release into Rolling

clalancette, 2021-11-08:
#903 was necessary because of the downstream change in https://github.com/ros2/rosidl/pull/627 .  Now that we've done a release of `rosidl` into Rolling, we need a new release of rosbag2 into Rolling to get it to build again: https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__rosbag2_cpp__ubuntu_focal_amd64__binary/204/ 

@emersonknapp Do you mind doing another new release here so that we can get `ros-rolling-desktop` working again?  Alternatively I'm happy to do it, just let me know.

emersonknapp, 2021-11-08:
I'm on vacation and won't be able to do it until at least next Monday (I only have my phone) - if you need it before then you'll be best off doing it

clalancette, 2021-11-08:
> I'm on vacation and won't be able to do it until at least next Monday (I only have my phone) - if you need it before then you'll be best off doing it

All right, happy to do it.  Thanks for the response!

jacobperron, 2022-01-13:
We can probably close this ticket since #907. But, I've also staged changes for a subsequent release in https://github.com/ros2/rosbag2/pull/948 :slightly_smiling_face:

closed 2022-01-25 by jacobperron