ros2/rosbag2 #909: ros2 bag record does not keep running under gdb in galactic

wjwwood, 2021-11-10:
## Description

When running `ros2 bag record ...` it will continue running until you stop it, e.g. with ctrl-c. However, in Galactic (but not in Rolling) when running under `gdb` (this is how I'm running it `gdb --args $(which ros2) bag record -a`) it will start normally but then immediately shutdown cleanly without any intervention. On Rolling it will run as expected until you stop it even under `gdb`.

### Expected Behavior

`gdb --args python3 $(which ros2) bag record -a` will continue to run until interrupted.

### Actual Behavior

Under Galactic it will exit cleanly as soon as it has finished subscribing (even if you don't use `-a`), but under Rolling it behaves as expected.

## To Reproduce

1. Source a Galactic installation
2. run `gdb --args python3 $(which ros2) bag record -a` and observe it exit without being stopped

## System (please complete the following information)

 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic
 - Version: `0.9.1`

## Additional context

Like I mentioned before, happens on Galactic but not in Rolling, so it must have been fixed. In another project we're using commit c387c58a1c38e914425dc0adf66dc022c63ceaef and it's broken there too (I believe this is between Galactic and Rolling).

wjwwood, 2021-11-10:
@jacobperron @ivanpauno FYI

wjwwood, 2021-11-10:
Example from Galactic:

```
william@ubuntu:/tmp
$ source /opt/ros/galactic/setup.bash
william@ubuntu:/tmp
$ ros2 bag record -a
1636580433.900229 [87]       ros2: using network interface enp0s5 (udp/10.211.55.6) selected arbitrarily from: enp0s5, docker0
[INFO] [1636580433.913127049] [rosbag2_storage]: Opened database 'rosbag2_2021_11_10-13_40_33/rosbag2_2021_11_10-13_40_33_0.db3' for READ_WRITE.
[INFO] [1636580433.913263699] [rosbag2_recorder]: Listening for topics...
[INFO] [1636580433.914343301] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1636580433.914885479] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
c^C[INFO] [1636580486.168307315] [rclcpp]: signal_handler(signal_value=2)
[INFO] [1636580486.190973021] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
william@ubuntu:/tmp
$ gdb --args python3 `which ros2` bag record -a
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...
(No debugging symbols found in python3)
(gdb) r
Starting program: /usr/bin/python3 /opt/ros/galactic/bin/ros2 bag record -a
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff536f700 (LWP 69504)]
1636580517.059543 [87]    python3: using network interface enp0s5 (udp/10.211.55.6) selected arbitrarily from: enp0s5, docker0
[New Thread 0x7ffff4b6e700 (LWP 69505)]
[New Thread 0x7fffeffff700 (LWP 69506)]
[New Thread 0x7fffef7fe700 (LWP 69507)]
[New Thread 0x7fffeeffd700 (LWP 69508)]
[New Thread 0x7fffee7fc700 (LWP 69509)]
[New Thread 0x7fffedffb700 (LWP 69510)]
[New Thread 0x7fffed7fa700 (LWP 69511)]
[New Thread 0x7fffecff9700 (LWP 69512)]
[INFO] [1636580517.080178978] [rosbag2_storage]: Opened database 'rosbag2_2021_11_10-13_41_57/rosbag2_2021_11_10-13_41_57_0.db3' for READ_WRITE.
[New Thread 0x7fffcffff700 (LWP 69514)]
[INFO] [1636580517.080683442] [rosbag2_recorder]: Listening for topics...
[INFO] [1636580517.081498084] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1636580517.081913974] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[New Thread 0x7fffcf7fe700 (LWP 69515)]
[New Thread 0x7fffceffd700 (LWP 69516)]
[Thread 0x7fffceffd700 (LWP 69516) exited]
[Thread 0x7fffcf7fe700 (LWP 69515) exited]
[INFO] [1636580517.183860222] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
[Thread 0x7fffcffff700 (LWP 69514) exited]
[Thread 0x7fffecff9700 (LWP 69512) exited]
[Thread 0x7fffed7fa700 (LWP 69511) exited]
[Thread 0x7fffedffb700 (LWP 69510) exited]
[Thread 0x7fffee7fc700 (LWP 69509) exited]
[Thread 0x7fffeeffd700 (LWP 69508) exited]
[Thread 0x7ffff4b6e700 (LWP 69505) exited]
[Thread 0x7fffeffff700 (LWP 69506) exited]
[Thread 0x7fffef7fe700 (LWP 69507) exited]
[Thread 0x7ffff536f700 (LWP 69504) exited]
[Inferior 1 (process 69491) exited normally]
(gdb) q
```

wjwwood, 2021-11-10:
Working example in Rolling:

```
william@ubuntu:~
$ source /opt/ros/rolling/setup.bash
william@ubuntu:~
$ cd /tmp/
william@ubuntu:/tmp
$ ros2 bag record -a
[INFO] [1636580197.652996532] [rosbag2_storage]: Opened database 'rosbag2_2021_11_10-13_36_37/rosbag2_2021_11_10-13_36_37_0.db3' for READ_WRITE.
[INFO] [1636580197.653120343] [rosbag2_recorder]: Listening for topics...
[INFO] [1636580197.653706559] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1636580197.654367812] [rosbag2_recorder]: Subscribed to topic '/rosout'
^C[INFO] [1636580203.731911817] [rclcpp]: signal_handler(signum=2)
[INFO] [1636580203.790304135] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
william@ubuntu:/tmp
$ gdb --args python3 `which ros2` bag record -a
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...
(No debugging symbols found in python3)
(gdb) r
Starting program: /usr/bin/python3 /opt/ros/rolling/bin/ros2 bag record -a
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff5793700 (LWP 66611)]
[New Thread 0x7ffff4f92700 (LWP 66612)]
[New Thread 0x7fffeffff700 (LWP 66613)]
[New Thread 0x7fffef7fe700 (LWP 66614)]
[New Thread 0x7fffeeffd700 (LWP 66615)]
[New Thread 0x7fffee7fc700 (LWP 66616)]
[New Thread 0x7fffedffb700 (LWP 66617)]
[New Thread 0x7fffed7fa700 (LWP 66618)]
[New Thread 0x7fffecff9700 (LWP 66619)]
[INFO] [1636580229.759900336] [rosbag2_storage]: Opened database 'rosbag2_2021_11_10-13_37_09/rosbag2_2021_11_10-13_37_09_0.db3' for READ_WRITE.
[New Thread 0x7fffcffff700 (LWP 66620)]
[INFO] [1636580229.760428375] [rosbag2_recorder]: Listening for topics...
[INFO] [1636580229.761303685] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1636580229.762001837] [rosbag2_recorder]: Subscribed to topic '/rosout'
[New Thread 0x7fffcf7fe700 (LWP 66621)]
^C
Thread 1 "python3" received signal SIGINT, Interrupt.
futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0xd28798) at ../sysdeps/nptl/futex-internal.h:183
183	../sysdeps/nptl/futex-internal.h: No such file or directory.
(gdb) q
A debugging session is active.

	Inferior 1 [process 66597] will be killed.

Quit anyway? (y or n) y
```

ivanpauno, 2021-11-11:
The issue is this:

https://github.com/ros2/rosbag2/blob/ecf831360a4b25639db1832b4eb5da95a5948e27/rosbag2_py/src/rosbag2_py/_transport.cpp#L194-L200

That code changed in https://github.com/ros2/rosbag2/pull/854, so the issue isn't present in galactic.
It's a fun problem that usually doesn't happen because there's a race: executor.cancel() **only** cancels a running spin, if it's called before spin() it does not cancel anything.

The current code in the `galactic` branch doesn't make any sense.
I guess someone copied https://github.com/ros2/rosbag2/blob/ecf831360a4b25639db1832b4eb5da95a5948e27/rosbag2_py/src/rosbag2_py/_transport.cpp#L132-L139 without noticing the difference between both cases.

ivanpauno, 2021-11-15:
Fixed by https://github.com/ros2/rosbag2/pull/911

closed 2021-11-15 by ivanpauno